gnuradio/pybombs #134: GQRX-digital will not launch in Ubuntu 14.04

n00b-research, 2015-01-02:
I have been having a problem with the launching of gqrx-digital in Ubuntu 14.04.  The installation will go just fine, specificaly after installing the dependencies for gnuradio through apt, then using pybombs, installed in the Home folder, to install gnuradio and then gqrx-digital.  However, after the installation I try to launch gqrx-digital from within the pybombs directory and receive the following error message:
"The program 'gqrx' is currently not installed. You can install it by typing:
sudo apt-get install gqrx-sdr"

After this I attempted to verify the gqrx-digital installation (sudo ./pybombs verify gqrx-digital), and this all checks out alright.  But, still, when I try to launch gqrx-digital I always get the same message.

Is there any direction that anyone can provide?

kantooon, 2015-01-07:
I'm not familiar with Ubuntu but what happens if you try to launch it directly from the terminal?

n00b-research, 2015-01-08:
That is what I have been doing, using the command "gqrx" from within the pybombs directory.  In Kali this is how gqrx is launched, when it is installed through pybombs.  In attempting to troubleshoot the problem I also have tried the command "gqrx-digital" from within pybombs.  Neither have worked, and both have led to the error cited above; 
"The program 'gqrx' is currently not installed. You can install it by typing:
sudo apt-get install gqrx-sdr"

n00b-research, 2015-01-08:
I should also mention that this install of Ubuntu was fresh, and that no other version of gnuradio was installed outside of using pybombs.  From what I have read up on thus far, having other versions installed can cause issues, at times, so I started with a fresh system.

kantooon, 2015-01-08:
Are you sure you know how to launch a local binary (one that is not in the system path)?

n00b-research, 2015-01-08:
I have been using pybombs for the better part of 2 months, on Kali,  and have always launched apps installed through pybombs in the same way, from the pybombs directory, through the name of the app.  So, to launch gunradio-companion, I always have typed "gnuradio-companion".  I have always launched gqrx in the same way, and even able to get gqrx-digital working in a version of Kali.  But, for some reason gqrx is not launching in this version of Ubuntu.  So, if you have launched applications in another way, and you feel that this would be helpful, please let me know.

n00b-research, 2015-01-08:
Figured it out.  In the past I had run pybombs with a target in /usr/local, and in this installation I installed in the Home folder.  After doing some more reading, specifically in this thread on Radio Reference, http://forums.radioreference.com/software-defined-radio/301459-virtualbox-project-run-op25-windows-7-a-2.html, it became clear that the primary difference was installation outside of a root folder.  When installing outside of a root folder it is important to run "source (location of setup_env.sh)" at the beginning of each session.

kantooon, 2015-01-08:
No, not really. You can just run the gqrx binary by using `./gqrx`. Also, there is no such thing as gqrx-digital, although it would probably be a good idea to change the name of the build target...

closed 2016-01-26 by mbr0wn