gnuradio/pybombs #153: Recipes fail to load on fresh intall of Ubuntu 14.04

wilsonbarret, 2015-02-14:
After a fresh install of Ubuntu 14.04, PyBombs fails to load any of the recipes.

mtc@mtc-lt:~/pybombs$ sudo ./pybombs install gnuradio
Settled on prefix: /home/mtc/target
Initializing environmental variables...
no existing inventory found, creating an empty one...
Loading recipes ...
Failed to load recipe libsndfile
Failed to load recipe gr-zmqblocks
Failed to load recipe gr-ieee-80211
Failed to load recipe gperftools
Failed to load recipe gr-mtb
Failed to load recipe pygtk
Failed to load recipe alsa
Failed to load recipe gr-osmosdr
Failed to load recipe pocsag-mrt
Failed to load recipe gr-ieee-802154
Failed to load recipe atk
Failed to load recipe gr-gsm
Failed to load recipe cheetah
Failed to load recipe bison
Failed to load recipe fecapi
Failed to load recipe gr-dvbs2
Failed to load recipe osmo-tetra
Failed to load recipe qwt5
Failed to load recipe gr-smithchart
Failed to load recipe gr-multimon
Failed to load recipe kal
Failed to load recipe rtl-sdr
Failed to load recipe hackrf
Failed to load recipe pmc
Failed to load recipe libpcap
Failed to load recipe pycairo
Failed to load recipe boost
Failed to load recipe libhidapi
Failed to load recipe swig
Failed to load recipe gr-ais
Failed to load recipe libusb
Failed to load recipe gnuradio
Failed to load recipe osmo-sdr
Failed to load recipe niusprio
Failed to load recipe gr-iqbal
Failed to load recipe gfortran
Failed to load recipe gr-radar
Failed to load recipe gr-ldpc
Failed to load recipe gr-packetradio
Failed to load recipe expat
Failed to load recipe bladeRF
Failed to load recipe gr-dsd
Failed to load recipe gr-fcdproplus
Failed to load recipe qt4
Failed to load recipe mpc
Failed to load recipe libosmo-dsp
Failed to load recipe theron
Failed to load recipe make
Failed to load recipe gr-fosphor
Failed to load recipe glfw3
Failed to load recipe gr-lte
Failed to load recipe cmake
Failed to load recipe pysstv
Failed to load recipe libpolarssl
Failed to load recipe ffi
Failed to load recipe gr-eventstream
Failed to load recipe mcpp
Failed to load recipe gnss-sdr
Failed to load recipe gr-rds
Failed to load recipe gtk2
Failed to load recipe pyqwt5
Failed to load recipe libxslt
Failed to load recipe mpfr
Failed to load recipe libbzip
Failed to load recipe gr-smartnet
Failed to load recipe airprobe
Failed to load recipe gr-dvbt
Failed to load recipe gqrx-digital
Failed to load recipe gr-tagutils
Failed to load recipe tornado
Failed to load recipe pygobject
Failed to load recipe gr-op25
Failed to load recipe gr-compat
Failed to load recipe gr-benchmark
Failed to load recipe lapack
Failed to load recipe apr
Failed to load recipe gr-dvbt2
Failed to load recipe cppunit
Failed to load recipe simple_radio_astronomy
Failed to load recipe flex
Failed to load recipe glib
Failed to load recipe gr-mediatools
Failed to load recipe wireshark
Failed to load recipe ice
Failed to load recipe fftw
Failed to load recipe git
Failed to load recipe python
Failed to load recipe pyqt4-dev-tools
Failed to load recipe libosmocore
Failed to load recipe gmp
Failed to load recipe gr-baz
Failed to load recipe libxml
Failed to load recipe numpy
Failed to load recipe gr-tutorial
Failed to load recipe airspy
Failed to load recipe gr-pocsag
Failed to load recipe pixman
Failed to load recipe libbtbb
Failed to load recipe scipy
Failed to load recipe gdk-pixbuf
Failed to load recipe sip
Failed to load recipe gr-pyqt
Failed to load recipe gr-specest
Failed to load recipe gr-extras
Failed to load recipe gr-drm
Failed to load recipe freetype
Failed to load recipe pango
Failed to load recipe gsl
Failed to load recipe gobject-introspection
Failed to load recipe cairo
Failed to load recipe armadillo
Failed to load recipe aprutil
Failed to load recipe fontconfig
Failed to load recipe wxpython
Failed to load recipe zeromq
Failed to load recipe gr-air-modes
Failed to load recipe db48
Failed to load recipe openlte
Failed to load recipe setuptools
Failed to load recipe gr-ettus
Failed to load recipe gtest
Failed to load recipe gqrx
Failed to load recipe glog
Failed to load recipe libpng
Failed to load recipe gr-theano
Failed to load recipe gr-pcap
Failed to load recipe gr-bluetooth
Failed to load recipe gr-ham
Failed to load recipe mbelib
Failed to load recipe log4cxx
Failed to load recipe graphviz
Failed to load recipe pyqt4
Failed to load recipe uhd
Failed to load recipe gr-elster
Failed to load recipe lxml
Failed to load recipe gr-as
Failed to load recipe gflags
unknown package gnuradio

miek, 2015-02-15:
I just ran into this one too, seems to have been introduced in 8aec423703383fb4f0776ccfc57987181f907549

The "Failed to load" error is from an exception in recipe_loader.py: "No option 'static' in section: 'config'"

I manually added "static = False" to my config.dat as a quick workaround and am building now.

osh, 2015-02-15:
this should be resolved now

closed 2015-02-15 by osh