gnuradio/pybombs #162: CMake Error at CMakeLists.txt:309 (message): VOLK required but not found.

YeLiu03, 2015-04-20:
Dear all,
install Gnuradio(gr-3.6) on ubuntul 12.04.5 using ./pybombs install gnuradio

gives:

CMake Error at CMakeLists.txt:309 (message):
  VOLK required but not found.

-- Configuring incomplete, errors occurred!
bash return val = 1
Traceback (most recent call last):
  File "./pybombs", line 124, in <module>
    install(p, not opts.force);
  File "/home/yeliu/Gnuradio/pybombs/mod_pybombs/pybombs_ops.py", line 101, in install
    global_recipes[pkg].install();
  File "/home/yeliu/Gnuradio/pybombs/mod_pybombs/recipe.py", line 537, in install
    st = self.install_src();
  File "/home/yeliu/Gnuradio/pybombs/mod_pybombs/recipe.py", line 602, in install_src
    self.install_order[step][1]();
  File "/home/yeliu/Gnuradio/pybombs/mod_pybombs/recipe.py", line 643, in configure
assert(st == 0);
AssertionError

Can you help me?
Regards,
YeLiu

osh, 2015-04-20:
Why not use 3.7?

YeLiu03, 2015-04-20:
@osh  Hello osh, I want to use Gnuradio+USRP N200 as a software defined RFID reader. 
The project owner use Gnu radio 3.6.2 (https://github.com/yqzheng/usrp2reader).
But I don't know if their source code can be applied to 3.7.

closed 2016-01-26 by mbr0wn