gnuradio/pybombs #166: Recipe qt4.lwr - source points to non-existent nokia ftp server

ptrkrysik, 2015-05-14:
Problem description in the title - the solution is to change source to:
source: wget://https://download.qt.io/archive/qt/4.6/qt-everywhere-opensource-src-4.6.2.tar.gz

ptrkrysik, 2015-05-15:
...after correcting the link the recipe still won't work:
```
./configure --prefix=/usr/local/ -opensource --confirm-license -L/usr/local//lib -L/usr/local//lib64 CC=gcc CXX=g++

Configuring: (100%) [==========================================================]
Configuration failed. Re-trying with higher verbosity.
make: *** No targets specified and no makefile found.  Stop.
Build failed. See output above for error messages.
```
The reason is CC=gcc CXX=g++ at the end of the configure command parameters - added automatically by Pybombs.

osh, 2015-05-15:
Is there a more verbose error of where this fails?
We could replace the configure stage with one that doesn't add CC and CXX for this call, but then it would break cross compiling --

ptrkrysik, 2015-05-16:
The configure script doesn't recognise CC and CXX variables as parameters and it prints help message when they are supplied by pybombs:
```
CC=gcc: unknown argument
CXX=g++: unknown argument

Usage:  configure [-h] [-prefix <dir>] [-prefix-install] [-bindir <dir>] [-libdir <dir>]
        [-docdir <dir>] [-headerdir <dir>] [-plugindir <dir> ] [-datadir <dir>]
        [-translationdir <dir>] [-sysconfdir <dir>] [-examplesdir <dir>]
        [-demosdir <dir>] [-buildkey <key>] [-release] [-debug]
        [-debug-and-release] [-developer-build] [-shared] [-static] [-no-fast] [-fast] [-no-largefile]
        [-largefile] [-no-exceptions] [-exceptions] [-no-accessibility]
        [-accessibility] [-no-stl] [-stl] [-no-sql-<driver>] [-sql-<driver>]
        [-plugin-sql-<driver>] [-system-sqlite] [-no-qt3support] [-qt3support]
        [-platform] [-D <string>] [-I <string>] [-L <string>] [-help]
        [-qt-zlib] [-system-zlib] [-no-gif] [-qt-gif] [-no-libtiff] [-qt-libtiff] [-system-libtiff]
        [-no-libpng] [-qt-libpng] [-system-libpng] [-no-libmng] [-qt-libmng]
        [-system-libmng] [-no-libjpeg] [-qt-libjpeg] [-system-libjpeg] [-make <part>]
        [-nomake <part>] [-R <string>]  [-l <string>] [-no-rpath]  [-rpath] [-continue]
        [-verbose] [-v] [-silent] [-no-nis] [-nis] [-no-cups] [-cups] [-no-iconv]
        [-iconv] [-no-pch] [-pch] [-no-dbus] [-dbus] [-dbus-linked]
        [-no-separate-debug-info] [-no-mmx] [-no-3dnow] [-no-sse] [-no-sse2]
        [-qtnamespace <namespace>] [-qtlibinfix <infix>] [-separate-debug-info] [-armfpa]
        [-no-optimized-qmake] [-optimized-qmake] [-no-xmlpatterns] [-xmlpatterns]
        [-no-multimedia] [-multimedia] [-no-phonon] [-phonon] [-no-phonon-backend] [-phonon-backend]
        [-no-audio-backend] [-audio-backend] [-no-openssl] [-openssl] [-openssl-linked]
        [-no-gtkstyle] [-gtkstyle] [-no-svg] [-svg] [-no-webkit] [-webkit] [-no-javascript-jit] [-javascript-jit]
        [-no-script] [-script] [-no-scripttools] [-scripttools] [-no-declarative] [-declarative]

        [additional platform specific options (see below)]
(...)
```

closed 2016-01-26 by mbr0wn