gnuradio/pybombs #188: Failed to install apache-thrift

nlthomas, 2015-09-02:
I successfully installed pybombs and am trying to install GNU Radio from the "app store."  I receive an error: "failed to install apache-thrift."  I proceeded to try to install apache-thrift myself using the instructions from: http://thrift.apache.org/docs/BuildingFromSource.  I am not the most efficient with ubunutu, and cannot seem to get ./bootstrap.h to run, but went ahead and ran the rest of the process and believe that I have apache-thrift installed because I am able to access the thrift --help.  I try using pybombs to again install gnu radio and I am still getting the same "failed to install apache-thrift" message.  Any ideas on how to proceed?

Thanks!

mdmbc, 2015-10-03:
If you edit config.dat and change satisfy_order = deb to satisfy_order = deb,src pybombs will pull the source and build it for you. Should install fine after that. It was failing for me because there is no deb option to install.

mayankjobanputra, 2015-10-04:
It worked for me!! Thank you.

closed 2015-10-04 by jmcorgan

giladbeeri, 2017-06-09:
GR 3.7.11.1 installed using the `gnuradio-stable` release, pybombs 2.3.0, Ubuntu 16.04. What worked for me was editing `$PREFIX/.pybombs/config.yml`, under `apache-thrift`, replacing `optional` to `forcebuild`, then running `pybombs install apache-thrift`.

Couldn't run `modtool` without apache-thrift (got exception saying `libthrift.so` is missing)