gnuradio/pybombs #189: "failed to install gcc" when installing with pybombs

jbarg, 2015-09-10:
When tying to install stuff via pybombs on Arch Linux following error occures. Any clue what to do?

```
pybombs(master)$ ./pybombs install gnuradio
Settled on prefix: /home/jon/sdr/target
Initializing environmental variables...
no existing inventory found, creating an empty one...
Loading recipes ...
/home/jon/sdr/target
Installing packages:
* make
* wget
* python
* libbzip
* boost
* fftw
* cppunit
* gcc
* swig
* gsl
* libusb
* curl
* git
* cmake
* mako
* numpy
* uhd
* alsa
* cheetah
* glib
* libtiff
* libpng
* libjpeg
* gdk-pixbuf
* atk
* freetype
* fontconfig
* pixman
* x11
* cairo
* pango
* gtk2
* wxpython
* libxml
* libxslt
* setuptools
* lxml
* ffi
* pycairo
* flex
* bison
* gobject-introspection
* pygobject
* pygtk
* sip
* qt4
* pyqt4
* qwt5
* pyqwt5
* libtool
* automake
* ssl
* libevent
* apache-thrift
* liblog4cpp
* gnuradio
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
[... 469 lines cut ...]
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install gcc
failed to install make
```

osh, 2015-09-10:
do you have gcc installed?

jbarg, 2015-09-10:
Yes, same error with clang.

frankcash, 2015-09-26:
I get a similar issue on OSX

ckuethe, 2015-10-26:
try "./pybombs -v -v -v install gnuradio 2>&1 | tee fail.txt"

Then you can see what's failing and how to correct it.

noc0lour, 2015-11-04:
I've run into a similar problem with gentoo. You should install gnuradio,gcc,wget,curl,git,cmake with your package manager or just make sure they are installed.
Then just edit config.dat and add them to forcepkgs =

qizhu8, 2015-11-14:
I come up with exactly the same problem. Have you found the solution? I tried to deal with the compilers, but failed.

closed 2016-01-26 by mbr0wn