gnuradio/pybombs #192: Can not install uhd with pybombs.   HELP

ahmednabeelma20001, 2015-09-22:
root@kali:~/Desktop/pybombs# ./pybombs install uhd
Settled on prefix: /usr/local
Initializing environmental variables...
Loading recipes ...
/usr/local
Installing packages:
* uhd
Installing from source: uhd
Building:    (100%) [==========================================================]
Build failed. Re-trying with reduced makewidth and higher verbosity.
[  1%] Built target getopt
[  5%] Built target man_page_gzips
[  6%] [  6%] [  6%] [  6%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_common.cpp.o
Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/db_xcvr2450.cpp.o
Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_version3.cpp.o
Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/db_rfx.cpp.o
virtual memory exhaustedvirtual memory exhaustedvirtual memory exhausted::  CCannot allocate memoryannot allocate memory:
 
Cannot allocate memory
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
lib/CMakeFiles/uhd.dir/build.make:2268: recipe for target 'lib/CMakeFiles/uhd.dir/usrp/dboard/db_rfx.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/dboard/db_rfx.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
lib/CMakeFiles/uhd.dir/build.make:2314: recipe for target 'lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_common.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/dboard/db_sbx_common.cpp.o] Error 1
lib/CMakeFiles/uhd.dir/build.make:2291: recipe for target 'lib/CMakeFiles/uhd.dir/usrp/dboard/db_xcvr2450.cpp.o' failed
make[2]: *** [lib/CMakeFiles/uhd.dir/usrp/dboard/db_xcvr2450.cpp.o] Error 1
CMakeFiles/Makefile2:156: recipe for target 'lib/CMakeFiles/uhd.dir/all' failed
make[1]: *** [lib/CMakeFiles/uhd.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
make: *** [all] Error 2
Build failed. See output above for error messages.
root@kali:~/Desktop/pybombs#

osh, 2015-09-22:
your VM ran out of memory, try one of ...
1. set makewidth=1 in your config.dat
2. increase your VM available memory

closed 2015-09-22 by osh

ahmednabeelma20001, 2015-09-22:
its not VM -- kali linux sana installed on hard drive

osh, 2015-09-22:
I suppose that eliminates option 2 then
if it won't build with makewidth=1 your machine simply doesnt have enough memory to build UHD regardless of pybombs