gnuradio/pybombs #193: sudo ./pybombs  install gnuradio  Fail!

greenoaktree, 2015-09-28:
sudo ./pybombs  install gnuradio
Settled on prefix: /usr/local
Initializing environmental variables...
no existing inventory found, creating an empty one...
Loading recipes ...
/usr/local
 fftw-devel not installed

This should be unreachable: in have_rpm()
broken package dependency gnuradio -> fftw (fftw recipe not found)

leftshift, 2015-10-22:
Same for me
I am running Elementary Freya 0.3.1 (Which is basically an Ubuntu 14.04)
It might also have to do with the fact that I kind of messed up my previous gnuradio installation

I'm going to further investigate this, alltough it failing in "have_rpm()" is a little odd, because it should be using debs all the way…

leftshift, 2015-10-23:
Okay, something goes wrong with sysutils.py:have_rpms()
My system reports to have rpms, but something after that goes wrong…
I'm going to take a look at ways for actually fixing this.

For now, you can just put
"return False"
in the first line of have_rpms() as a quick hack.
This of course will prevent it from looking for rpms, which at least in my case is not a problem

greenoaktree, 2015-10-26:
thanks,I will try this way

closed 2016-01-26 by mbr0wn