gnuradio/pybombs #195: Error compiling apache-thrift when adding "--depth=1" to gitoptions

pachulo, 2015-10-13:
In the configuration phase of pybombs I added this git option:

    Options to always pass to git clones (--depth=1 or --single-branch will speed up clones)
    gitoptions []:--depth=1

But when trying to install apache-thrift I got this error:

    Installing from source: apache-thrift
    Cloning into 'apache-thrift'...
    remote: Counting objects: 2259, done.
    remote: Compressing objects: 100% (1839/1839), done.
    remote: Total 2259 (delta 773), reused 955 (delta 332), pack-reused 0
    Receiving objects: 100% (2259/2259), 2.76 MiB | 874.00 KiB/s, done.
    Resolving deltas: 100% (773/773), done.
    Checking connectivity... done.
    error: pathspec 'e2e5e22c1' did not match any file(s) known to git.

Because the commit 'e2e5e22c1' is not in the downloaded git repository history.

jmcorgan, 2015-10-13:
This is expected, as the specific commit is not at the tip of a branch, and thus wouldn't be part of a depth=1 clone.

pachulo, 2015-10-14:
Ok; shouldn't a warning be shown when setting the options then?

closed 2016-01-26 by mbr0wn