gnuradio/pybombs #200: apache-thrift error

vsboost, 2015-11-13:
Hi all,

Trying to install gnuradio via pybombs and it craps out when installing apache-thrift, any ideas?

---------------------------------------------------------
make[4]: Entering directory '/home/tom/pybombs/src/apache-thrift/lib/cpp'
/bin/bash ../../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift  -I/usr/include  -I./src -D__STDC_LIMIT_MACROS -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5  -fPIC  -Wall -Wextra -pedantic -DNDEBUG -std=c++11 -MT src/thrift/qt/libthriftqt5_la-moc__TQTcpServer.lo -MD -MP -MF src/thrift/qt/.deps/libthriftqt5_la-moc__TQTcpServer.Tpo -c -o src/thrift/qt/libthriftqt5_la-moc__TQTcpServer.lo `test -f 'src/thrift/qt/moc__TQTcpServer.cpp' || echo './'`src/thrift/qt/moc__TQTcpServer.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift -I/usr/include -I./src -D__STDC_LIMIT_MACROS -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I/usr/include/x86_64-linux-gnu/qt5 -fPIC -Wall -Wextra -pedantic -DNDEBUG -std=c++11 -MT src/thrift/qt/libthriftqt5_la-moc__TQTcpServer.lo -MD -MP -MF src/thrift/qt/.deps/libthriftqt5_la-moc__TQTcpServer.Tpo -c src/thrift/qt/moc__TQTcpServer.cpp  -fPIC -DPIC -o src/thrift/qt/.libs/libthriftqt5_la-moc__TQTcpServer.o
src/thrift/qt/moc__TQTcpServer.cpp:13:2: error: #error "This file was generated using the moc from 4.8.6. It"
 #error "This file was generated using the moc from 4.8.6. It"
  ^
src/thrift/qt/moc__TQTcpServer.cpp:14:2: error: #error "cannot be used with the include files from this version of Qt."
 #error "cannot be used with the include files from this version of Qt."
  ^
src/thrift/qt/moc__TQTcpServer.cpp:15:2: error: #error "(The moc has changed too much.)"
 #error "(The moc has changed too much.)"
  ^
src/thrift/qt/moc__TQTcpServer.cpp:61:7: error: 'QMetaObjectExtraData' does not name a type
 const QMetaObjectExtraData apache::thrift::async::TQTcpServer::staticMetaObjectExtraData = {
       ^
src/thrift/qt/moc__TQTcpServer.cpp:63:2: warning: extra ';' [-Wpedantic]
 };
  ^
src/thrift/qt/moc__TQTcpServer.cpp:67:57: error: 'staticMetaObjectExtraData' was not declared in this scope
       qt_meta_data_apache__thrift__async__TQTcpServer, &staticMetaObjectExtraData }
                                                         ^
src/thrift/qt/moc__TQTcpServer.cpp: In member function 'virtual const QMetaObject* apache::thrift::async::TQTcpServer::metaObject() const':
src/thrift/qt/moc__TQTcpServer.cpp:76:71: error: conditional expression between distinct pointer types 'QDynamicMetaObjectData*' and 'const QMetaObject*' lacks a cast
     return QObject::d_ptr->metaObject ? QObject::d_ptr->metaObject : &staticMetaObject;
                                                                       ^
src/thrift/qt/moc__TQTcpServer.cpp:77:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:1351: recipe for target 'src/thrift/qt/libthriftqt5_la-moc__TQTcpServer.lo' failed
make[4]: *** [src/thrift/qt/libthriftqt5_la-moc__TQTcpServer.lo] Error 1
make[4]: Leaving directory '/home/tom/pybombs/src/apache-thrift/lib/cpp'
Makefile:1574: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/tom/pybombs/src/apache-thrift/lib/cpp'
Makefile:538: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/tom/pybombs/src/apache-thrift/lib'
Makefile:609: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/tom/pybombs/src/apache-thrift'
Makefile:530: recipe for target 'all' failed
make: *** [all] Error 2
Build failed. See output above for error messages.

vsboost, 2015-11-14:
i changed the install path from home/name to root and it seems to work now, so looks like something wrong in my home path??

ckuethe, 2015-11-17:
I'm curious about why thrift is picking up some broken files. I'm doing basically the same thing, building from ~/pybombs.

If you feel like debugging it further, could you capture the full compile output ("./pybombs install -v -v -v apache-thrift") and post it as a gist?

vsboost, 2015-11-17:
Thanks ckuethe

 

I think I’ll leave it as is for now, it took me a while to get gnuradio going again and I’m afraid of breaking it lol

ckuethe, 2015-11-17:
Part of the reason I suggest installing as non-root into a non-standard
location (like someplace in your home directory) is so that you don't break
a working system. So you can have ~/pybombs-target-00 ..
~/pybombs-target-99 and none of them will collide. Additionally, once you
get it all working, you can make a backup archive (zip/tar) so that if
something does corrupt your installation it's easy to restore.

But I'm glad that it's working for you now.

closed 2016-01-05 by ckuethe

marwon34, 2019-02-03:
make  all-recursive
make[1]: Entering directory '/root/pybombs/bladeRF/src/apache-thrift'
Making all in lib/cpp
make[2]: Entering directory '/root/pybombs/bladeRF/src/apache-thrift/lib/cpp'
Making all in .
make[3]: Entering directory '/root/pybombs/bladeRF/src/apache-thrift/lib/cpp'
/bin/bash ../../libtool  --tag=CXX   --mode=compile g++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift -I../../lib/c_glib/src/thrift  -I/usr/include  -I./src -D__STDC_LIMIT_MACROS   -Wall -Wextra -pedantic -DNDEBUG -MT src/thrift/transport/libthriftz_la-TZlibTransport.lo -MD -MP -MF src/thrift/transport/.deps/libthriftz_la-TZlibTransport.Tpo -c -o src/thrift/transport/libthriftz_la-TZlibTransport.lo test -f 'src/thrift/transport/TZlibTransport.cpp' || echo './'src/thrift/transport/TZlibTransport.cpp
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I../../lib/cpp/src/thrift -I../../lib/c_glib/src/thrift -I/usr/include -I./src -D__STDC_LIMIT_MACROS -Wall -Wextra -pedantic -DNDEBUG -MT src/thrift/transport/libthriftz_la-TZlibTransport.lo -MD -MP -MF src/thrift/transport/.deps/libthriftz_la-TZlibTransport.Tpo -c src/thrift/transport/TZlibTransport.cpp  -fPIC -DPIC -o src/thrift/transport/.libs/libthriftz_la-TZlibTransport.o
In file included from src/thrift/transport/TZlibTransport.cpp:23:
./src/thrift/transport/TZlibTransport.h:26:10: fatal error: zlib.h: No such file or directory
 #include <zlib.h>
          ^~~~~~~~
compilation terminated.
make[3]: * [Makefile:1463: src/thrift/transport/libthriftz_la-TZlibTransport.lo] Error 1
make[3]: Leaving directory '/root/pybombs/bladeRF/src/apache-thrift/lib/cpp'
make[2]: * [Makefile:1693: all-recursive] Error 1
make[2]: Leaving directory '/root/pybombs/bladeRF/src/apache-thrift/lib/cpp'
make[1]: * [Makefile:646: all-recursive] Error 1
make[1]: Leaving directory '/root/pybombs/bladeRF/src/apache-thrift'
make: * [Makefile:560: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package apache-thrift:
Build failed.

marwon34, 2019-02-03:
I use Kali for Linux 2019.1