gnuradio/pybombs #201: gr-osmosdr enable nonfree for sdrplay

vsboost, 2015-11-14:
Any way to -DENABLE_NON_FREE=TRUE when gr-osmosdr is being built? i am using an sdrplay and apparently this flag needs to be called for it to work.

Cheers

ringof, 2015-11-15:
I manually added the line:
 var config_opt = " -DENABLE_NONFREE=TRUE "
in the recipe file gr-osmosdr.lwr in pybombs/recipe/, before the "inherit: cmake" line.

vsboost, 2015-11-15:
I tried that, still cant pick up the sdrplay in gqrx or gnuradio, any other suggestions.

vsboost, 2015-11-15:
Ok so my bad, its -DENABLE_NONFREE=TRUE not -DENABLE_NON_FREE=TRUE

All good now

Thanks for that.. :-)

closed 2015-11-15 by vsboost

wcat13, 2016-03-20:
I am trying to rebuild gr-osmosdr with var config_opt = " -DENABLE_NONFREE=TRUE " added just above the "inherit: cmake" line. in ~/wayne/.pybombs/recipes/gr-recipes/gr-osmosdr.lwr

```
pybombs rebuild gr-osmosdr
yaml.scanner.ScannerError: while scanning a simple key
  in "<string>", line 32, column 1:
    var config_opt = " -DENABLE_NONF ... 
    ^
could not found expected ':'
  in "<string>", line 33, column 1:
    inherit: cmake
    ^
```
Hoping one of you can tell me what I am doing wrong?
Regards

ringof, 2016-03-20:
Just went through this last week. They changed up how the options are
written (booo PyBOMBS docs!). I found this to work:

var: {config_opt: ' -DENABLE_NONFREE=TRUE'}

wcat13, 2016-03-20:
Fantastic.

Thank you!

mbr0wn, 2016-03-21:
PyBOMBS docs need improvement (which happens by providing patches), but I would like point out that all files are in YAML format, which the documentation does emphasize.

ringof, 2016-03-21:
I am going to find an appropriate home for this tidbit of configuration information.