gnuradio/pybombs #202: gqrx-issue

vsboost, 2015-11-19:
Probably not a pyboms issue but i thought i would report it and hopefully get a solution.  This is on a new kali installation with the stock gnuradio removed.

kali:~/pybombs# ./pybombs install gqrx -vvv
Settled on prefix: /root/target
Initializing environmental variables...
---------- loading recipes -------------------
Loading recipes ...
Loading recipes ... done
---------- loading recipes finished ----------

Checking if gqrx is installed: No
/root/target
TMPFILE = /root/target/tmpzAGKv2
WRITE PERMS OK /root/target/tmpzAGKv2
gqrx dep [['gnuradio', 'gr-osmosdr', 'pulseaudio', 'qt4', 'git']]
gnuradio dep [['make', 'boost', 'fftw', 'cppunit', 'swig', 'gsl', 'uhd', 'alsa', 'git', 'python', 'cheetah', 'wxpython', 'numpy', 'lxml', 'pygtk', 'pycairo', 'cmake', 'pyqt4', 'pyqwt5', 'gcc', 'apache-thrift', 'liblog4cpp', 'git', 'cmake']]
gr-osmosdr dep [['uhd', 'rtl-sdr', 'osmo-sdr', 'hackrf', 'gnuradio', 'gr-iqbal', 'bladeRF', 'airspy', 'git', 'cmake']]
pulseaudio dep [[]]
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of libpulse-dev (5.0) is >= than 1.1
qt4 dep [['wget']]
PyBombs.sysutils - INFO - have_deb: Satisfies requirement...installed version of libqt4-dev (4.8.6) is >= than 4.6.2
git dep [['wget', 'curl']]
Installing packages:
* gqrx
install type priority: ['src']
install src called (gqrx)
Installing from source: gqrx
[('fetch', <bound method recipe.fetch of <mod_pybombs.recipe.recipe instance at 0x7f0d5e42a3f8>>), ('configure', <bound method recipe.configure of <mod_pybombs.recipe.recipe instance at 0x7f0d5e42a3f8>>), ('make', <bound method recipe.make of <mod_pybombs.recipe.recipe instance at 0x7f0d5e42a3f8>>), ('installed', <bound method recipe.installed of <mod_pybombs.recipe.recipe instance at 0x7f0d5e42a3f8>>)]
Current step: (gqrx :: make)
make
('\n    make -j4\n', '\n    make -j$makewidth\n')
('\n    make -j4\n', '\n    make -j4\n')
bash exec (/root/pybombs/src/gqrx):: make -j4
g++ -m64 -Wl,-O1 -o gqrx main.o mainwindow.o receiver.o remote_control.o remote_control_settings.o cafsk12.o costabf.o agc_impl.o correct_iq_cc.o decimator.o hbf_decim.o lpf.o decoder_impl.o parser_impl.o resampler_xx.o rx_agc_xx.o rx_demod_am.o rx_demod_fm.o rx_fft.o rx_filter.o rx_meter.o rx_noise_blanker_cc.o rx_rds.o sniffer_f.o stereo_demod.o udp_sink_f.o afsk1200win.o agc_options.o audio_options.o bookmarks.o bookmarkstablemodel.o bookmarkstaglist.o demod_options.o dockaudio.o dockbookmarks.o dockinputctl.o dockrds.o dockrxopt.o dockfft.o freqctrl.o ioconfig.o iq_tool.o meter.o nb_options.o plotter.o qtcolorpicker.o nbrx.o receiver_base.o wfmrx.o pa_device_list.o pa_sink.o pa_source.o moc_mainwindow.o moc_remote_control.o moc_remote_control_settings.o moc_cafsk12.o moc_afsk1200win.o moc_agc_options.o moc_audio_options.o moc_bookmarks.o moc_bookmarkstablemodel.o moc_bookmarkstaglist.o moc_demod_options.o moc_dockaudio.o moc_dockbookmarks.o moc_dockfft.o moc_dockinputctl.o moc_dockrds.o moc_dockrxopt.o moc_freqctrl.o moc_ioconfig.o moc_iq_tool.o moc_meter.o moc_nb_options.o moc_plotter.o moc_qtcolorpicker.o qrc_icons.o qrc_textfiles.o    -L/usr/lib/x86_64-linux-gnu -lboost_system -lboost_program_options -lrt -lpulse-simple -lpulse -L/root/target/lib -lgnuradio-analog -lvolk -lgnuradio-digital -lgnuradio-filter -lgnuradio-fft -lgnuradio-osmosdr -lgnuradio-blocks -lgnuradio-runtime -lgnuradio-pmt -lQtGui -lQtNetwork -lQtCore -lpthread 
/usr/bin/ld: warning: libvolk.so.1.1git, needed by /root/target/lib/libgnuradio-analog.so, may conflict with libvolk.so.0.0.0
/usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `uhd::device::find(uhd::device_addr_t const&)'
collect2: error: ld returned 1 exit status
Makefile:257: recipe for target 'gqrx' failed
make: *** [gqrx] Error 1
bash return val = 2
Build failed. Re-trying with reduced makewidth and higher verbosity.
make
('\n    make -j4\n', '\n    make -j$makewidth\n')
('\n    make -j4\n', '\n    make -j4\n')
bash exec (/root/pybombs/src/gqrx):: make -j4
g++ -m64 -Wl,-O1 -o gqrx main.o mainwindow.o receiver.o remote_control.o remote_control_settings.o cafsk12.o costabf.o agc_impl.o correct_iq_cc.o decimator.o hbf_decim.o lpf.o decoder_impl.o parser_impl.o resampler_xx.o rx_agc_xx.o rx_demod_am.o rx_demod_fm.o rx_fft.o rx_filter.o rx_meter.o rx_noise_blanker_cc.o rx_rds.o sniffer_f.o stereo_demod.o udp_sink_f.o afsk1200win.o agc_options.o audio_options.o bookmarks.o bookmarkstablemodel.o bookmarkstaglist.o demod_options.o dockaudio.o dockbookmarks.o dockinputctl.o dockrds.o dockrxopt.o dockfft.o freqctrl.o ioconfig.o iq_tool.o meter.o nb_options.o plotter.o qtcolorpicker.o nbrx.o receiver_base.o wfmrx.o pa_device_list.o pa_sink.o pa_source.o moc_mainwindow.o moc_remote_control.o moc_remote_control_settings.o moc_cafsk12.o moc_afsk1200win.o moc_agc_options.o moc_audio_options.o moc_bookmarks.o moc_bookmarkstablemodel.o moc_bookmarkstaglist.o moc_demod_options.o moc_dockaudio.o moc_dockbookmarks.o moc_dockfft.o moc_dockinputctl.o moc_dockrds.o moc_dockrxopt.o moc_freqctrl.o moc_ioconfig.o moc_iq_tool.o moc_meter.o moc_nb_options.o moc_plotter.o moc_qtcolorpicker.o qrc_icons.o qrc_textfiles.o    -L/usr/lib/x86_64-linux-gnu -lboost_system -lboost_program_options -lrt -lpulse-simple -lpulse -L/root/target/lib -lgnuradio-analog -lvolk -lgnuradio-digital -lgnuradio-filter -lgnuradio-fft -lgnuradio-osmosdr -lgnuradio-blocks -lgnuradio-runtime -lgnuradio-pmt -lQtGui -lQtNetwork -lQtCore -lpthread 
/usr/bin/ld: warning: libvolk.so.1.1git, needed by /root/target/lib/libgnuradio-analog.so, may conflict with libvolk.so.0.0.0
/usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `uhd::device::find(uhd::device_addr_t const&)'
collect2: error: ld returned 1 exit status
Makefile:257: recipe for target 'gqrx' failed
make: *** [gqrx] Error 1
bash return val = 2
Build failed. See output above for error messages.

ckuethe, 2015-11-19:
Thanks for including the "-vvv" output! That makes the problem easy to see. And the problem is...

     -L/usr/lib/x86_64-linux-gnu ... -L/root/target/lib

This instructs the linker to first look in /usr/lib, then in /root/target/lib. It picked up libgnuradio-osmosdr from /usr/lib but it didn't find an acceptable libuhd (which should probably be a dependency of osmosdr) which I would assume is in /usr/lib also.

You could try editing the Makefile and adding "-luhd" to the list of libraries linked. I'd also suggest uninstalling any other gnuradio/osmocom/etc related packages from the system.

vsboost, 2015-11-19:
thanks for that, issuing: "apt-get remove gr-osmosdr" did the trick

closed 2015-11-19 by vsboost

stackprogramer, 2016-09-23:
g