gnuradio/pybombs #203: Error while trying to install gnuradio 3.6 - untracked working files

strasharo, 2015-12-21:
I'm trying to install gnu-radio 3.6 on latest Kali, so i inserted the following line in the recipe:

gitrev: tags/v3.6.5.1

Everything goes smooth until the point where it checks out gnuradio and tries to switch branches:

kali@kali:~/pybombs$ sudo ./pybombs install gnuradio
Settled on prefix: /usr/local
Initializing environmental variables...
Loading recipes ...
/usr/local
Installing packages:
* gnuradio
Installing from source: gnuradio
fatal: destination path 'gnuradio' already exists and is not an empty directory.
Cloning into 'gnuradio'...
Checking connectivity... done.
Cloning into 'volk'...
remote: Counting objects: 5631, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 5631 (delta 4), reused 0 (delta 0), pack-reused 5616
Receiving objects: 100% (5631/5631), 1.61 MiB | 379.00 KiB/s, done.
Resolving deltas: 100% (4020/4020), done.
Checking connectivity... done.
error: The following untracked working tree files would be overwritten by checkout:
	volk/CMakeLists.txt
	volk/apps/CMakeLists.txt
	volk/apps/volk_profile.cc
	volk/cmake/msvc/config.h
	volk/cmake/msvc/inttypes.h
	volk/cmake/msvc/stdint.h
	volk/gen/archs.xml
	volk/gen/machines.xml
	volk/gen/volk_arch_defs.py
	volk/gen/volk_compile_utils.py
	volk/gen/volk_kernel_defs.py
	volk/gen/volk_machine_defs.py
	volk/gen/volk_tmpl_utils.py
	volk/include/volk/volk_common.h
	volk/include/volk/volk_complex.h
	volk/include/volk/volk_prefs.h
	volk/lib/CMakeLists.txt
	volk/lib/qa_utils.cc
	volk/lib/qa_utils.h
	volk/lib/testqa.cc
	volk/lib/volk_prefs.c
	volk/lib/volk_rank_archs.c
	volk/lib/volk_rank_archs.h
	volk/tmpl/volk.tmpl.c
	volk/tmpl/volk.tmpl.h
	volk/tmpl/volk_config_fixed.tmpl.h
	volk/tmpl/volk_cpu.tmpl.c
	volk/tmpl/volk_cpu.tmpl.h
	volk/tmpl/volk_machine_xxx.tmpl.c
	volk/tmpl/volk_machines.tmpl.c
	volk/tmpl/volk_machines.tmpl.h
	volk/tmpl/volk_typedefs.tmpl.h
	volk/volk.pc.in
Please move or remove them before you can switch branches.
Aborting
error: The following untracked working tree files would be overwritten by checkout:
	volk/CMakeLists.txt
	volk/apps/CMakeLists.txt
	volk/apps/volk_profile.cc
	volk/cmake/msvc/config.h
	volk/cmake/msvc/inttypes.h
	volk/cmake/msvc/stdint.h
	volk/gen/archs.xml
	volk/gen/machines.xml
	volk/gen/volk_arch_defs.py
[... 290 lines cut ...]
	volk/gen/volk_machine_defs.py
	volk/gen/volk_tmpl_utils.py
	volk/include/volk/volk_common.h
	volk/include/volk/volk_complex.h
	volk/include/volk/volk_prefs.h
	volk/lib/CMakeLists.txt
	volk/lib/qa_utils.cc
	volk/lib/qa_utils.h
	volk/lib/testqa.cc
	volk/lib/volk_prefs.c
	volk/lib/volk_rank_archs.c
	volk/lib/volk_rank_archs.h
	volk/tmpl/volk.tmpl.c
	volk/tmpl/volk.tmpl.h
	volk/tmpl/volk_config_fixed.tmpl.h
	volk/tmpl/volk_cpu.tmpl.c
	volk/tmpl/volk_cpu.tmpl.h
	volk/tmpl/volk_machine_xxx.tmpl.c
	volk/tmpl/volk_machines.tmpl.c
	volk/tmpl/volk_machines.tmpl.h
	volk/tmpl/volk_typedefs.tmpl.h
	volk/volk.pc.in
Please move or remove them before you can switch branches.
Aborting
kali@kali:~/pybombs$

ckuethe, 2015-12-21:
Try removing all the offending files, or check out the files by hand... Why are you using 3.6.5?

strasharo, 2015-12-23:
I tried removing those offending files manually, however it checks them there again and does the same thing. 
Was trying to get topguw running, but since 3.6 doesn't build using pybombs I ended up using the zmiana patch for airprobe.

closed 2016-01-26 by mbr0wn