gnuradio/pybombs #209: Problem with libboost1.53-dev

jsa91, 2016-01-13:
When I try to install gr-gsm through pybombs the following problem appears: 

"PyBombs.sysutils - WARNING - deb_exists: could not find a downloadable version of libboost1.53-dev
Installing from source: boost"

The installation gets aborted and fails.
I'm pretty new to Linux and has hoping someone could help me with this.

Thanks!

ckuethe, 2016-01-13:
You didn't say which distribution or version you're using.

jsa91, 2016-01-13:
Sorry, I'm using Kali Linux 2.0

jsa91, 2016-01-13:
Pybombs through "git clone" so I assume the latest version. I'm using Kali Linux 2.0.

ckuethe, 2016-01-14:
OK, I haven't tested with Kali. I'd approach fixing it by looking at what
version of boost development libraries are available in the repositories
and adjusting the deb satisfier to match.

ckuethe, 2016-01-25:
Just tested - PyBOMBS works on a fresh install Kali 2016.01 which provides Boost 1.58. Try updating with "apt-get update && apt-get dist-upgrade" before running PyBOMBS.

closed 2016-01-25 by ckuethe