gnuradio/pybombs #221: Make better use of git cache?

mbr0wn, 2016-01-26:
[image] **Issue by [ckuethe](https://github.com/ckuethe)**
_Thursday Jul 30, 2015 at 05:31 GMT_
_Originally opened as https://github.com/gnuradio/pybombs2/issues/9_

----

If a git reference directory is configured in the global git options (perhaps because the user ran `src-reference.sh`), fetch should check to see if the recipe's repo is cached in the reference directory. If it's missing, fetch can add it, then do a local fetch from there.

mbr0wn, 2016-01-26:
[image] **Comment by [mbr0wn](https://github.com/mbr0wn)**
_Thursday Jul 30, 2015 at 16:02 GMT_

----

I like it. @sdh11 is working on the fetchers, so I'll let him comment. However, this is an optimization, which I personally do not care about *at all* having in there right now.

mbr0wn, 2016-01-26:
[image] **Comment by [ckuethe](https://github.com/ckuethe)**
_Thursday Jul 30, 2015 at 16:32 GMT_

----

The use case is for big repositories like gnuradio, UHD, or lte-cell-scanner... Or anything from osmocom. 

Actually, when I think about it, defaulting to a reference directory would be a good choice.

mbr0wn, 2016-01-26:
[image] **Comment by [mbr0wn](https://github.com/mbr0wn)**
_Thursday Jul 30, 2015 at 16:45 GMT_

----

Just to be clear: This is a great thing to have, no discussions there, I just don't have any cycles to spare right now.

mbr0wn, 2016-01-26:
[image] **Comment by [sdh11](https://github.com/sdh11)**
_Thursday Jul 30, 2015 at 17:30 GMT_

----

I agree. This would be a good optimization for later.

mbr0wn, 2016-11-20:
We already have the `git-cache` option. Just need a good way to set up a cache.

jmcorgan, 2016-11-20:
When testing/releasing the livesdr build I might do 20 builds in a row.  There is ccache for the compiles, squid for the packages, and pybombs1 had the git-cache setup. Would love to have this now for the livesdr-pybombs2 branch.

mbr0wn, 2016-11-22:
I started working on it. It's not entirely trivial what to cache, and when, but I think I have a sane flow figured out. Now I just need to magic time to implement it.

mbr0wn, 2016-11-24:
c65db67d04bb737b4eb6b6f4631849ebf7f6f8b5

Barring bugs, that should do it.

closed 2016-11-24 by mbr0wn