gnuradio/pybombs #240: PyBOMBS version parse error in new install of Fedora 23 64 bit?

ettermyra, 2016-01-29:
Running Python 2.7.10. Installed PyBOMBS 2.0.0 using pip 8.0.2 through sudo. I've set basic config for preferred compiler, git user, preferred prefix and such.
However, when running the command: `pybombs -v install gnuradio` I get:
```
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - Parsing config file failed (/etc/pybombs/config.yml).
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/user/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'cc': 'gcc', 'cxx': 'g++', 'default_prefix': 'mygnuradio', 'gituser': 'user'}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/lib/python2.7/site-packages/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/user/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias mygnuradio.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/user/system/gnuradio)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/user/system/gnuradio/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix configuration file not found: /home/user/system/gnuradio/.pybombs/config.yml, assuming empty.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/user/system/gnuradio/src
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix inventory file not found: /home/user/system/gnuradio/.pybombs/inventory.yml
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/user/system/gnuradio/.pybombs/inventory.yml...
PyBombs.Inventory - DEBUG - No success. Creating empty inventory.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/user/system/gnuradio/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/user/system/gnuradio/.pybombs/recipes', '/home/user/.pybombs/recipes/gr-etcetera', '/home/user/.pybombs/recipes/gr-recipes', '/usr/lib/python2.7/site-packages/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/user/system/gnuradio/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/user/system/gnuradio/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/user/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/user/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 50 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/user/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/user/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 159 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/lib/python2.7/site-packages/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 5 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.install - DEBUG - Initializing command class for command install
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - yumdnf is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'yumdnf', 'pkgconfig', 'cmd', 'source']
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - yumdnf is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'yumdnf', 'pkgconfig', 'cmd', 'source']
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Checking if package gnuradio is installed.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
[... 496 lines cut ...]
  File "/usr/lib/python2.7/site-packages/pybombs/recipe.py", line 53, in ev
    return func(self.name, self.compare or ">=", self.version)
  File "/usr/lib/python2.7/site-packages/pybombs/packagers/yum.py", line 72, in _package_installed
    return vcompare(comparator, installed_version, required_version)
  File "/usr/lib/python2.7/site-packages/pybombs/utils/vcompare.py", line 34, in vcompare
    return operators[cmp_op](StrictVersion(version_x), StrictVersion(version_y))
  File "/usr/lib64/python2.7/distutils/version.py", line 40, in __init__
    self.parse(vstring)
  File "/usr/lib64/python2.7/distutils/version.py", line 107, in parse
    raise ValueError, "invalid version number '%s'" % vstring
ValueError: invalid version number '1'
```
It seem to have trouble finding out the version of make, so when running command `make -v` I get:
```
GNU Make 4.0
Built for x86_64-redhat-linux-gnu
Copyright (C) 1988-2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
Any ideas?

Best regards
/T

mbr0wn, 2016-01-30:
What's the output of 'dnf list installed make'?

ettermyra, 2016-01-30:
Output:
```
Installed Packages
make.x86_64       1:4.0-5.1.fc23
```
Quick assessment; pybombs is not taking the `1:` into consideration when looking for the version string? Just a guess as I haven't looked at the pybombs code yet.

mbr0wn, 2016-01-30:
Yes that's likely what's happening. Can you submit a fix for this? I don't have a Fedora box anywhere right now. You'll need to edit line 134 in yum.py to include an optional leading digit and colon.

ettermyra, 2016-01-30:
Preliminary results show that modifying the regexp to handle the `x:` (x as in any future version number) works. I'll do some more testing to see that the fix is solid.

mbr0wn, 2016-01-30:
Yeah not all packages have a x: prefix, so it has to work with both but still return the right number. Looks like you're on the right track though :+1:

mbr0wn, 2016-02-04:
Does #246 fix this for you?

mbr0wn, 2016-02-18:
Assume this was fixed.

closed 2016-02-18 by mbr0wn