gnuradio/pybombs #241: Running "python setup.py install" as an unprivileged user does not clean up completely after failure

JonZeolla, 2016-01-30:
Running "python setup.py install" as an unprivileged user does not clean up completely after failure.  I was working on a fresh Ubuntu 14.04 VM using snapshots to revert as appropriate.  I would expect a run of python setup.py install as an unprivileged user would clean itself up properly, so that you could run it as root immediately after the failure with success.

**Fresh install, accidentally running setup.py as unprivileged user**
> **sdr@ubuntu:~/pybombs$ python setup.py install**
> Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-18.4.zip
> Extracting in /tmp/tmpFe8NEX
> Now working in /tmp/tmpFe8NEX/setuptools-18.4
> Building a Setuptools egg in /home/sdr/pybombs
> /home/sdr/pybombs/setuptools-18.4-py2.7.egg
> running install
> error: can't create or remove files in install directory
> 
> The following error occurred while trying to add or remove files in the
> installation directory:
> 
>     [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/test-easy-install-4827.write-test'
> 
> The installation directory you specified (via --install-dir, --prefix, or
> the distutils default setting) was:
> 
>     /usr/local/lib/python2.7/dist-packages/
> 
> Perhaps your account does not have write access to this directory?  If the
> installation directory is a system-owned directory, you may need to sign in
> as the administrator or "root" account.  If you do not have administrative
> access to this machine, you may wish to choose a different installation
> directory, preferably one that is listed in your PYTHONPATH environment
> variable.
> 
> For information on other options, you may wish to consult the
> documentation at:
> 
>   https://pythonhosted.org/setuptools/easy_install.html
> 
> Please make the appropriate changes for your system and try again.
> 
> **sdr@ubuntu:~/pybombs$ python setup.py install**
> Traceback (most recent call last):
>   File "setup.py", line 72, in <module>
>     from setuptools import setup
>   File "build/bdist.linux-x86_64/egg/setuptools/__init__.py", line 12, in <module>
>   File "build/bdist.linux-x86_64/egg/setuptools/extension.py", line 8, in <module>
>   File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 21, in <module>
> AttributeError: 'module' object has no attribute 'packaging'
> **sdr@ubuntu:~/pybombs$ sudo python setup.py install**
> Traceback (most recent call last):
>   File "setup.py", line 72, in <module>
>     from setuptools import setup
>   File "build/bdist.linux-x86_64/egg/setuptools/__init__.py", line 12, in <module>
>   File "build/bdist.linux-x86_64/egg/setuptools/extension.py", line 8, in <module>
>   File "build/bdist.linux-x86_64/egg/setuptools/dist.py", line 21, in <module>
> AttributeError: 'module' object has no attribute 'packaging'

**Fresh install, running setup.py as root the first time around**

> **sdr@ubuntu:~/pybombs$ sudo python setup.py install**
> Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-18.4.zip
> Extracting in /tmp/tmpnEyofA
> Now working in /tmp/tmpnEyofA/setuptools-18.4
> Building a Setuptools egg in /home/sdr/pybombs
> /home/sdr/pybombs/setuptools-18.4-py2.7.egg
> running install
> running bdist_egg
> running egg_info
> creating PyBOMBS.egg-info
> writing requirements to PyBOMBS.egg-info/requires.txt
> writing PyBOMBS.egg-info/PKG-INFO
> writing top-level names to PyBOMBS.egg-info/top_level.txt
> writing dependency_links to PyBOMBS.egg-info/dependency_links.txt
> writing entry points to PyBOMBS.egg-info/entry_points.txt
[... 286 lines cut ...]
> Downloading https://pypi.python.org/packages/source/P/PyYAML/PyYAML-3.11.zip#md5=89cbc92cda979042533b640b76e6e055
> Processing PyYAML-3.11.zip
> Writing /tmp/easy_install-t7f_3C/PyYAML-3.11/setup.cfg
> Running PyYAML-3.11/setup.py -q bdist_egg --dist-dir /tmp/easy_install-t7f_3C/PyYAML-3.11/egg-dist-tmp-nYj3Aa
> build/temp.linux-x86_64-2.7/check_libyaml.c:2:18: fatal error: yaml.h: No such file or directory
>  #include <yaml.h>
>                   ^
> compilation terminated.
> 
> libyaml is not found or a compiler error: forcing --without-libyaml
> (if libyaml is installed correctly, you may need to
>  specify the option --include-dirs or uncomment and
>  modify the parameter include_dirs in setup.cfg)
> zip_safe flag not set; analyzing archive contents...
> Moving PyYAML-3.11-py2.7-linux-x86_64.egg to /usr/local/lib/python2.7/dist-packages
> Adding PyYAML 3.11 to easy-install.pth file
> 
> Installed /usr/local/lib/python2.7/dist-packages/PyYAML-3.11-py2.7-linux-x86_64.egg
> Searching for requests==2.2.1
> Best match: requests 2.2.1
> Adding requests 2.2.1 to easy-install.pth file
> 
> Using /usr/lib/python2.7/dist-packages
> Finished processing dependencies for PyBOMBS==2.0.0
>

mbr0wn, 2016-01-30:
This looks like an issue with distutils / setuptools. Or do you have any suggestion how to fix this? It's not like the setup.py file is anything special.

JonZeolla, 2016-02-02:
Not sure, just wanted to bring it up to you so that you were aware.

closed 2016-02-02 by JonZeolla