gnuradio/pybombs #243: Issue with github tag v2.0.0 - please update versions appropriately

JonZeolla, 2016-01-31:
It appears that there is a bug with the tag v2.0.0 which doesn't exist in the current version of the pybombs github repo (even though the current version of the pybombs github repo results in 2.0.0 when I run `pybombs --version`).  

When I run `pybombs install gqrx`, I get the following error if I cloned pybombs by using `git clone --recursive --branch v2.0.0 https://github.com/gnuradio/pybombs -q`:
> ...
> PyBombs.RecipeListManager - ERROR - Package None has no recipe file!
> sdr@ubuntu:~$ pybombs --version
> 2.0.0

If I run `git clone --recursive https://github.com/gnuradio/pybombs -q` it installs properly and I get the following result when I look at the pybombs version:
> sdr@ubuntu:~$ pybombs --version
> 2.0.0

mbr0wn, 2016-01-31:
The tag got lost when we moved the repos. I added it back. Does this solve your issue?

JonZeolla, 2016-01-31:
It seems to be better now - thanks.

closed 2016-01-31 by JonZeolla