gnuradio/pybombs #244: PyBombs installation failure

lvp, 2016-02-01:
I'm trying to install gnuradio using pyBombs. I've tried this on several different laptops but, they all fail. Even with a clean install of UBUNTU 14.04 LTS it does not come to completion. I've added the terminal output.

Anyone else the same problem?

regards, 
Laurent van Poppelen

[pybombs_install_error.txt](https://github.com/gnuradio/pybombs/files/112242/pybombs_install_error.txt)

mbr0wn, 2016-02-01:
@lvp thanks -- this is a pretty major bug and I can't believe it's in there. gr-iqbal has gnuradio as dependency, but it doesn't install it first.

mbr0wn, 2016-02-04:
@lvp also I answered the email you sent me, but your mail server rejected my response.

mbr0wn, 2016-02-18:
This should be good now.

closed 2016-02-18 by mbr0wn