gnuradio/pybombs #249: Installation failure on OpenSuSe 42.1

drom, 2016-02-04:
```
drom@drom:~/work> pip install PyBOMBS
Collecting PyBOMBS
  Using cached PyBOMBS-2.0.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-ujeo9glo/PyBOMBS/setup.py", line 73, in <module>
        import pybombs
      File "/tmp/pip-build-ujeo9glo/PyBOMBS/pybombs/__init__.py", line 24, in <module>
        from utils import dict_merge
    ImportError: No module named 'utils'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ujeo9glo/PyBOMBS
```

mbr0wn, 2016-02-21:
I'd guess this doesn't happen anymore. Can you confirm?

    pip install --upgrade git+https://github.com/gnuradio/pybombs.git

drom, 2016-02-21:
Yes. It works. Thank you.

closed 2016-02-21 by drom