gnuradio/pybombs #251: setup.py initial failure on Ubuntu-based systems

JonZeolla, 2016-02-07:
I've tested this on Xubuntu, Lubuntu, and Ubuntu 14.04 and 15.10.  Below is specifically from a fresh install of Ubuntu 15.10:

`sudo apt-get install git cmake libboost-all-dev gnuradio-dev sox`
`sudo apt-get update`
`sudo apt-get upgrade`
`git clone --recursive https://github.com/gnuradio/pybombs`
`cd pybombs`

[image]

If I simply run `sudo python setup.py install` a second time, it succeeds.

mbr0wn, 2016-02-21:
Was not able to reproduce this on a vanilla 14.04 VM.

JonZeolla, 2016-02-21:
This appears to be partially fixed.  Using the same commands as before, here is what I saw:

**Ubuntu 14.04 Desktop** completes successfully the first time ($? == 0), but on the first run there is a non-critical yaml.h error.  If I run a second time, I do not see this error.
[image]
[image]

**Ubuntu 14.04 Server** completes successfully the first time ($? == 0), but on the first run there is a non-critical yaml.h error.  If I run a second time, I do not see this error.
[image]

**Ubuntu 15.10 Desktop** does not complete successfully the first time.  If I re-run `setup.py install` I get the non-critical yaml.h error, and if I run it a third time it looks clean.  
[image]
[image]

**Ubuntu 15.10 Server** does not complete successfully the first time.  If I re-run `setup.py install` I get the non-critical yaml.h error, and if I run it a third time it looks clean.  
[image]

mbr0wn, 2016-02-21:
Does

    $ sudo pip install git+https://github.com/gnuradio/pybombs

throw the same error?

mbr0wn, 2016-02-21:
OK, git it now. There's an incompatibility with `ez_setup.py` and newer setuptools. I can apply a workaround, maybe, but it won't be a great solution, until someone fixes `ez_setup.py` (not me!).

mbr0wn, 2016-02-21:
Can't come up with a good workaround. At least, I improved the error message. FYI, `pip install` should always work if that's any help for you. Closing this as there's not much more I can do right now.

closed 2016-02-21 by mbr0wn

mbr0wn, 2016-02-22:
I found a newer version of ez_setup which seems to fix the issue.

JonZeolla, 2016-02-22:
Ok great, I assume that was 2a588d5.  I'll test later today

JonZeolla, 2016-02-22:
Confirmed - this seems to be resolved.  Thanks!