gnuradio/pybombs #256: pyboms inconsitency after refetch

keesj, 2016-02-17:
While doing a install of bladeRF (after a forces refetch) the build system thinks
that bladeRF is already configured while this is clearly not the case(bladeRF never installed properly) so there is a state inconsitency

```
PyBombs.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (bladeRF)
PyBombs.PackageManager - DEBUG - Using packager source
PyBombs.Packager.source - DEBUG - State on package bladeRF is 30
PyBombs.Packager.source - DEBUG - Package bladeRF is already fetched.
PyBombs.Packager.source - DEBUG - Using build directory: /home/keesj/projects/sdr/target/src/bladeRF/host/build
PyBombs.Packager.source - DEBUG - Package bladeRF is already configured.
PyBombs.Packager.source - DEBUG - Package bladeRF is already built.
PyBombs.Packager.source - DEBUG - Installing package bladeRF
PyBombs.Packager.source - DEBUG - In cwd - /home/keesj/projects/sdr/target/src/bladeRF/host/build
PyBombs.monitor_process() - DEBUG - Executing command `make install'
make: *** No rule to make target 'install'.  Stop.
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 2
PyBombs.Packager.source - ERROR - Problem occured while building package bladeRF:
Installation failed
PyBombs.install - ERROR - Error installing package bladeRF. Aborting.
```

mbr0wn, 2016-02-18:
@keesj with the other fixes, can you see if this is still an issue?

keesj, 2016-02-18:
@mbr0wn 

I did a full rm -rf of target  and initiated a A full build and this worked this time:
```
time pybombs -vvv install gnuradio gr-osmosdr 2>&1 | tee out.log
/home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/main.py
real    30m50.480s
user    98m52.672s
sys     4m15.148s
```

A pyboms refresh bladeRF also kinda looks ok:
```
Fetching origin
Your branch is up-to-date with 'origin/master'.
HEAD is now at 1b96a9c Updated CONTRIBUTORS file
```

a pybombs rebuild however still does not work:
```
keesj@keesj-Lenovo-Y50-70:~$ pybombs rebuild bladeRF
PyBombs.rebuild - ERROR - Package bladeRF is not installed into current prefix. Aborting.
```

While bladeRF was insrtalled (as dependency of an other component)

Content of my inventory:
```
apache-thrift:
  source: git+https://github.com/apache/thrift.git
  state: 40
bladeRF:
  source: git+https://github.com/Nuand/bladeRF.git
  state: 20
gnuradio:
  source: git+https://github.com/gnuradio/gnuradio.git
  state: 40
gr-iqbal:
  source: git+git://git.osmocom.org/gr-iqbal
  state: 40
libosmo-dsp:
  source: git+git://git.osmocom.org/libosmo-dsp
  state: 40
osmo-sdr:
  source: git+git://git.osmocom.org/osmo-sdr.git
  state: 40
rtl-sdr:
  source: git+git://git.osmocom.org/rtl-sdr
  state: 40
uhd:
  source: git+https://github.com/EttusResearch/uhd.git
  state: 40

```

I can therefore not rebuild bladeRF

mbr0wn, 2016-02-18:
Yup, I think I fixed this.

closed 2016-02-19 by keesj