gnuradio/pybombs #258: pyboms install bladeRF ignores INSTALL_UDEV_RULES=OFF

keesj, 2016-02-17:
https://github.com/gnuradio/gr-recipes/blob/master/bladeRF.lwr#L34 configured bladeRF to not install udev rules but this option somehow does not get passed to cmake (or does not have the desired result)

```
keesj@keesj-Lenovo-Y50-70:~$ pybombs -vvv install bladeRF
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - Parsing config file failed (/etc/pybombs/config.yml).
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/keesj/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'main'}
PyBombs.ConfigManager - DEBUG - Template directory: /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/keesj/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias main.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/keesj/projects/sdr/target)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/keesj/projects/sdr/target/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/keesj/projects/sdr/target/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/keesj/projects/sdr/target/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/keesj/projects/sdr/target/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/keesj/projects/sdr/target/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/keesj/projects/sdr/target/.pybombs/recipes', '/home/keesj/.pybombs/recipes/gr-etcetera', '/home/keesj/.pybombs/recipes/gr-recipes', '/home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/keesj/projects/sdr/target/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/keesj/projects/sdr/target/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/sdk.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/setuptools.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bootstrapautoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/autoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/empty.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/boostflags.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/distutils.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bjam.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/cmake.lwt
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/keesj/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/keesj/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/sdk.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/setuptools.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bootstrapautoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/autoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/empty.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/boostflags.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/distutils.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bjam.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/cmake.lwt
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/keesj/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/keesj/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/sdk.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/setuptools.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bootstrapautoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/autoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/empty.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/boostflags.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/distutils.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bjam.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/cmake.lwt
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 5 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/sdk.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/setuptools.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/bootstrapautoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/autoconf.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/empty.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/boostflags.lwt
PyBombs.RecipeListManager - OBNOXIOUS - Adding template /home/keesj/projects/sdr/pybombs/build/lib.linux-x86_64-2.7/pybombs/templates/distutils.lwt
[... cut ...]
Call Stack (most recent call first):
  misc/cmake_install.cmake:37 (include)
  cmake_install.cmake:38 (include)

Makefile:85: recipe for target 'install' failed
make: *** [install] Error 1
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 2
PyBombs.Packager.source - ERROR - Problem occured while building package bladeRF:
Installation failed
PyBombs.install - ERROR - Error installing package bladeRF. Aborting.
keesj@keesj-Lenovo-Y50-70:~$ 

```

keesj, 2016-02-17:
In the log you can see:

```
configure: ./configure --prefix=$prefix $config_opt CC=$cc CXX=$cxx
depends: []
gitargs: ''
gitbranch: master
gitrev: ''
inherit: false
install: make install
installdir: .
make: make -j$makewidth
satisfy: {cmd: git --version, deb: git-core || git, rpm: git}
source: ['wget+http://git-core.googlecode.com/files/git-1.7.9.6.tar.gz']
target: package
uninstall: make uninstall
vars: {config_opt: ''}
```
e.g. config_opt becomes empty

mbr0wn, 2016-02-18:
That was actually a typo in the recipe, caused by a bug in our converter script. Fixed the bug in 4daf5728a2ea9326c2769e8f6450123252c4758a and updated the recipe.

closed 2016-02-18 by mbr0wn