gnuradio/pybombs #260: AttributeError: StrictVersion instance has no attribute 'version'

wright3189, 2016-02-19:
I'm getting this error no matter what I try to install with PyBOMBS:

[garrett@localhost ~]$ uname -a
Linux localhost.localdomain 4.3.5-300.fc23.x86_64 #1 SMP Mon Feb 1 03:18:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
[garrett@localhost ~]$ sudo pybombs -v install gnuradio
[sudo] password for garrett: 
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - Parsing config file failed (/etc/pybombs/config.yml).
PyBombs.ConfigManager - DEBUG - Reading config info from file: /root/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'default_prx'}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /root/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using CWD as prefix (/home/garrett)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/garrett/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix configuration file not found: /home/garrett/.pybombs/config.yml, assuming empty.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/garrett/src
PyBombs.ConfigManager.PrefixInfo - DEBUG - Source dir does not exist.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix inventory file not found: /home/garrett/.pybombs/inventory.yml
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/garrett/.pybombs/inventory.yml...
PyBombs.Inventory - DEBUG - No success. Creating empty inventory.
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/root/.pybombs/recipes/gr-etcetera', '/root/.pybombs/recipes/gr-recipes', '/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /root/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/root/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /root/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/root/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 5 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.install - DEBUG - Initializing command class for command install
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - yumdnf is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'yumdnf', 'pkgconfig', 'cmd', 'source']
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - yumdnf is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'yumdnf', 'pkgconfig', 'cmd', 'source']
PyBombs.PackageManager - DEBUG - Checking if package gnuradio is installed.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Checking if package gnuradio is installed.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Checking if package gnuradio is installed.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
PyBombs.PackageManager - DEBUG - Package gnuradio is requesting a source build.
[... 470 lines cut ...]
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/dep_manager.py", line 85, in make_tree_recursive
    subtree = self.make_tree_recursive(dep, filter_callback)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/dep_manager.py", line 85, in make_tree_recursive
    subtree = self.make_tree_recursive(dep, filter_callback)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/dep_manager.py", line 76, in make_tree_recursive
    if not self.pm.exists(pkg):
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/package_manager.py", line 113, in exists
    pkg_version = pkgr.exists(r)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/packagers/base.py", line 56, in exists
    return self._packager_run_tree(recipe, self._package_exists)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/packagers/base.py", line 132, in _packager_run_tree
    return satisfy_rule.ev(satisfy_evaluator)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/recipe.py", line 74, in ev
    return self.first.ev(func) or self.second.ev(func)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/recipe.py", line 53, in ev
    return func(self.name, self.compare or ">=", self.version)
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/packagers/yum.py", line 59, in _package_exists
    if required_version is not None and not vcompare(comparator, available_version, required_version):
  File "/usr/lib/python2.7/site-packages/PyBOMBS-2.0.0-py2.7.egg/pybombs/utils/vcompare.py", line 34, in vcompare
    return operators[cmp_op](StrictVersion(version_x), StrictVersion(version_y))
  File "/usr/lib64/python2.7/distutils/version.py", line 140, in __cmp__
    compare = cmp(self.version, other.version)
AttributeError: StrictVersion instance has no attribute 'version'

Any ideas on the issue or how to fix it? I'm on the latest version of Fedora, as show above.

mbr0wn, 2016-02-19:
This error is coming from within distutils. That's weird. Which version distutils do you have?

wright3189, 2016-02-19:
These are the module versions I could find:

[root@localhost garrett]# python
Python 2.7.10 (default, Sep  8 2015, 17:20:17) 
[GCC 5.1.1 20150618 (Red Hat 5.1.1-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import distutils
>>> distutils.__version__
'2.7.10'
>>> 
[root@localhost garrett]# pip freeze
chardet==2.2.1
iniparse==0.4
kitchen==1.2.1
pexpect==4.0.1
ptyprocess==0.5
PyBOMBS==2.0.0
pycurl==7.19.5.1
pygobject==3.18.2
pygpgme==0.3
pyliblzma==0.5.3
pyxattr==0.5.3
pyxdg==0.25
PyYAML==3.11
requests==2.9.1
rpm-python==4.13.0rc1
six==1.9.0
urlgrabber==3.10.1
yum-metadata-parser==1.1.4

mbr0wn, 2016-02-19:
Hm I have the same distutils version. Also, it's specific to the gdk-pixbuf package... This'll take a bit before I can look into this; I'll probably need a Fedora VM or something.

mbr0wn, 2016-02-19:
Can you patch your yum.py file like this, and then re-run with `-vvv`:

```{.py}
 74     def _package_install(self, pkgname, comparator=">=", required_version=None, cmd='install'):
 75         """
 76         Call 'COMMAND install pkgname' if we can satisfy the version requirements.
 77         """
 78         available_version = self.get_available_version_from_pkgr(pkgname)
 79         self.log.obnoxious("{0} has version {1}".format(pkgname, available_version))
 80         if required_version is not None and not vcompare(comparator, available_version, required_version):
 81             return False                                        
 82         try:
 83             subproc.monitor_process([self.command, "-y", cmd, pkgname], elevate=True)
 84             return True
 85         except Exception as ex:
 86             self.log.error("Running `{0} install' failed.".format(self.command))
 87             self.log.obnoxious(str(ex))
 88         return False
```

Thanks!

mbr0wn, 2016-02-19:
Note line 79, with the logging. That's what's new.

mbr0wn, 2016-02-20:
This error would occur if an empty string is passed to a StrictVersion.

mbr0wn, 2016-02-20:
Got hold of a Fedora VM and I can't reproduce this. I recommend updating your PyBOMBS and recipes, maybe even starting from scratch, and trying again. A couple of things were changed, for once, the StrictVersion was replaced by a LooseVersion, but that wouldn't change this.

wright3189, 2016-02-20:
I'm still getting the same error as before. I attached the output for you to view.
[PYBOMBS.txt](https://github.com/gnuradio/pybombs/files/138918/PYBOMBS.txt)

wright3189, 2016-02-20:
I was using a fresh install of Fedora on a VM. Did you dnf update? Only difference might be is that I'm using a LXDE spin.

mbr0wn, 2016-02-20:
@wright3189 You're not using the latest version though. `pip install --upgrade git+<URL>`

wright3189, 2016-02-20:
Yeah, after updating, I'm getting:
AttributeError: LooseVersion instance has no attribute 'version'

Going to try again with a vanilla copy of Fedora

mbr0wn, 2016-02-20:
OK, your log tipped me off. It's quite a massive bug, actually. However, I have no idea why I'm not running into it. Try 25fa8ce893c95b5a04b5faf2c095309f77bd7e8e, that should FTFY.

mbr0wn, 2016-02-20:
@wright3189 if this actually fixes it for you, please close this issue.

wright3189, 2016-02-20:
Didn't try the patch yet. Decided to start with a fresh install of Fedora, everything updated. At first I got this:

CMake Error at CMakeLists.txt:69 (message):
  LibUSB 1.0 required to compile rtl-sdr

After installing libusb_devel I was able to get it going again but then ran into this:

PyBombs.install - ERROR - Error installing package apache-thrift. Aborting.

wright3189, 2016-02-20:
[apache-thrift-error.txt](https://github.com/gnuradio/pybombs/files/139283/apache-thrift-error.txt)

This is the error I receive with pybombs -vvv install apache-thrift

Thanks,
Andrew

mbr0wn, 2016-02-21:
There's a known issue with thrift and Fedora. If you set apache-thrift to forceinstalled, it will skip its installation and GNU Radio will simply not have control port. There's nothing we can do from here, so I'm closing this issue.

closed 2016-02-21 by mbr0wn