gnuradio/pybombs #261: Boost pybombs error

Pradyumna007, 2016-02-19:
PyBombs.install - INFO - Installing package: boost
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libboost-dev : Depends: libboost1.54-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
PyBombs.Packager.source - WARNING - Build dir already exists: /home/vprady/Documents/myprefix/src/boost/build
Installing:  (100%) [=========================================================]
PyBombs.Packager.source - ERROR - Problem occured while building package boost:
Installation failed
PyBombs.install - ERROR - Error installing package boost. Aborting.

mbr0wn, 2016-02-19:
The error message is telling you that your local apt-get state is broken. This is not a PyBOMBS issue.

closed 2016-02-19 by mbr0wn

Pradyumna007, 2016-02-20:
How to change the path of myprefix in pybombs2.0 version .... i actually gave it in Documents/myprefix folder ..... bt when i am trying to run gnuradio-companion, my system is not executing. How to solve this .... how to change the prefix and also how to copy all downloaded and installed files in Documents/myprefix to make them run ....

mbr0wn, 2016-02-22:
Please ask questions like these on the discuss-gnuradio mailing list.