gnuradio/pybombs #262: No CMAKE_CXX_COMPILER could be found.

wright3189, 2016-02-21:
I Forceinstalled apache-thrift, but now something with CMAKE_CXX_COMPILER.

PyBombs.monitor_process() - DEBUG - Executing command `CC= CXX= cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/pybombs  -Wno-dev'
Configuring: (100%) [=======================================================================================================]
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 1
PyBombs.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
PyBombs.Packager.source - DEBUG - Configuring recipe uhd
PyBombs.Packager.source - DEBUG - Using vars - {'config_opt': ''}
PyBombs.Packager.source - DEBUG - In cwd - /pybombs/src/uhd/host/build
PyBombs.monitor_process() - DEBUG - Executing command `CC= CXX= cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/pybombs  -Wno-dev'
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:27 (PROJECT):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/pybombs/src/uhd/host/build/CMakeFiles/CMakeOutput.log".
See also "/pybombs/src/uhd/host/build/CMakeFiles/CMakeError.log".
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 1
PyBombs.Packager.source - ERROR - Configuration failed after running at least twice.
PyBombs.Packager.source - ERROR - Problem occurred while building package uhd:
Configuration failed
PyBombs.install - ERROR - Error installing package uhd. Aborting.

Any tips?

wright3189, 2016-02-21:
Looks like all I needed was gcc-c++. Installation finished successfully.

closed 2016-02-21 by wright3189

mbr0wn, 2016-02-22:
We could make build essentials a dependency of the source package manager.