gnuradio/pybombs #268: Make build essentials a dependency of the source package manager

mbr0wn, 2016-02-22:
See also #262.

Notes:
- On Debian and Ubuntu, there's the build-essential package
- Other distros may have this too
- Otherwise, include gcc, make
- Put the .lwr into the main repo, not gr-recipes

mbr0wn, 2016-02-27:
88abdd8124cd8b6fc02173123dba0a977a2dd130

closed 2016-02-27 by mbr0wn

peseta3, 2016-12-15:
Hallo together,
I have the same problem.

> PyBombs.Packager.source - ERROR - Configuration failed after running at least twice.
> PyBombs.Packager.source - ERROR - Problem occurred while building package uhd:
> Configuration failed
> PyBombs.install - ERROR - Error installing package uhd. Aborting.

But I don`t know what you say with this

> On Debian and Ubuntu, there's the build-essential package
> Otherwise, include gcc, make
> Put the .lwr into the main repo, not gr-recipes

Please help me
Thank you