gnuradio/pybombs #273: CMake build from src doesn't get pointed to target pgkconfigdir for dependencies built from src

awalls-cx18, 2016-02-28:
When building a package from source (e.g. gr-fosphor), PyBOMBS (2.x) does not let CMake know about the ${prefix}/lib/pkgconfig directory, where pkg-config, used by CMake's FindPkgConfig module, can find package config files for dependencies PyBOMBS built from source (e.g. glfw3 for gr-phosphor).

In my particular case, the CMake build of gr-fosphor was (luckily?) finding the header file for the glfw3 built from source and installed under the ${prefix}, but the library found was the system's installed glfw2 library.  A snippet from CMakeCache.txt from a broken gr-fosphor build:

//Path to a file.
GLFW3_INCLUDE_DIRS:PATH=/home/andy/target/include     <--- Correct
//Path to a library.
GLFW3_LIBRARIES:FILEPATH=/usr/lib/libglfw.so   <---- Incorrect

The solution was to inform pkg-config about the target installation path in an environment variable before running PyBOMBS to build gr-fosphor:

PKG_CONFIG_PATH=${prefix}/lib/pkgconfig

which adds a path for pkg-config to search first, before all the default pkg-config paths.

It seems like PyBOMBS should be setting this automatically for at least CMake builds.
From the documentations, there was no obvious way, to me, to set the environment variable in the ~/.pybombs/config.yml file to have an env: resolve using the current ${prefix} in use.

mbr0wn, 2016-02-29:
Good point -- I'm pondering if I should simply 'source' setup_env.sh before launching any process, like we're sourcing the environment file. That would set `PKG_CONFIG_PATH`.

A short term fix for you would be to amend your config.yml as such:

```{.yml}
[env]
  PKG_CONFIG_PATH: /path/to/prefix/lib/pkgconfig:$PKG_CONFIG_PATH
```

mbr0wn, 2016-02-29:
...and you can use `$PYBOMBS_PREFIX` as an env variable.

awalls-cx18, 2016-02-29:
Yeah.  Both of those workarounds seemed to break the PyBOMBS2 management of multiple builds stored under different prefixes though.

mbr0wn, 2016-03-03:
You can edit the config.yml file inside the prefix's .pybombs dir.

u0m3, 2016-03-10:
It is my impression that it ignores the `env` section or at least it does not export the vars properly.
I ran into this issue myself and my work-around was to export the env vars by hand.

My **~/.pybombs/config.yaml**
```yaml
--- 
categories: 
  common: 
    forcebuild: true
config: 
  cc: gcc
  cxx: g++
  default_prefix: default
  makewidth: "8"
  satisfy_order: "native,src"
env: 
  LD_LIBRARY_CONFIG: "~/gnuradio-env/lib:$LD_LIBRARY_CONFIG"
  LD_LIBRARY_PATH: "~/gnuradio-env/lib:$LD_LIBRARY_PATH"
  PATH: "~/gnuradio-env/bin:$PATH"
  PKG_CONFIG_PATH: "~/gnuradio-env/lib/pkgconfig:$PKG_CONFIG_PATH"
  PYTHONPATH: "~/gnuradio-env/lib/python2.7/dist-packages:$PYTHONPATH"
packages: 
  gnuradio: 
    forcebuild: true
    forceinstalled: false
  uhd:
    forcebuild: true
    forceinstalled: false
  gr-fcdproplus:
    forcebuild: true
    forceinstalled: false
prefix_aliases: 
  default: ~/gnuradio-env
recipes: 
  gr-etcetera: /home/rpopescu/.pybombs/recipes/gr-etcetera
  gr-recipes: /home/rpopescu/.pybombs/recipes/gr-recipes
```

mbr0wn, 2016-03-12:
@u0m3 env sections inside config.yml files should be read.

    pybombs prefix env

will print the environment as its used for commands.

@awalls-cx18 do you mind giving f93023d0461319d0e5c1ab44bc1cace8cd1eda7b a try? It creates a default environment based on the prefix.

mbr0wn, 2016-03-12:
...it's on master.

awalls-cx18, 2016-03-12:
Tried the latest on master.  It appears to work. :)

mbr0wn, 2016-03-12:
Cool!

closed 2016-03-12 by mbr0wn