gnuradio/pybombs #277: rtl-sdr: error installing in PyBOMBS due to missing source dir

gareth8118, 2016-03-10:
I have run into the following issue whilst trying to install `gr-osmosdr` using PyBOMBS:

```
$ pybombs -p gnuradio install rtl-sdr
|
\- rtl-sdr
PyBombs.install - INFO - Installing package: rtl-sdr
PyBombs.Packager.source - ERROR - Problem occurred while building package rtl-sdr:
There should be a source dir in /home/gareth/gnuradio/src/rtl-sdr, but there isn't.
PyBombs.install - ERROR - Error installing package rtl-sdr. Aborting.
```
PyBOMBS is pip installed from github. This is a newly minted Ubuntu 15.10 installation that I'm putting gnuradio onto.

I can't tell what the root cause of this issue is, even with verbose output - is it a bug in PyBOMBS or something I have wrong locally?:

```
$ pybombs -v -p gnuradio install rtl-sdr
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/gareth/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/local/lib/python2.7/dist-packages/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/gareth/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias gnuradio.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing prefix dir from command line: /home/gareth/gnuradio
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/gareth/gnuradio/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/gareth/gnuradio/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/gareth/gnuradio/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/gareth/gnuradio/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/gareth/gnuradio/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/gareth/gnuradio/.pybombs/recipes', '/home/gareth/.pybombs/recipes/gr-etcetera', '/home/gareth/.pybombs/recipes/gr-recipes', '/usr/local/lib/python2.7/dist-packages/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/gareth/gnuradio/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/gareth/gnuradio/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/gareth/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/gareth/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/gareth/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/gareth/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/local/lib/python2.7/dist-packages/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/local/lib/python2.7/dist-packages/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 6 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.install - DEBUG - Initializing command class for command install
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - apt-get is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'apt-get', 'pkgconfig', 'cmd', 'source']
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - apt-get is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'apt-get', 'pkgconfig', 'cmd', 'source']
PyBombs.PackageManager - DEBUG - Checking if package rtl-sdr is installed.
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.PackageManager - DEBUG - Checking if package rtl-sdr is installed.
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.PackageManager - DEBUG - Checking if package rtl-sdr is installed.
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.PackageManager - DEBUG - Checking if package libusb is installed.
PyBombs.ReqScanner - DEBUG - Parsing 'libusb-1.0-0-dev >= 1.0.9'
PyBombs.Packager.apt-get - DEBUG - Package libusb-1.0-0-dev has version 1.0.19 in dpkg
PyBombs.PackageManager - DEBUG - Checking if package cmake is installed.
PyBombs.ReqScanner - DEBUG - Parsing '(cmake >= 2.8.3) && (cmake-data >= 2.8.3)'
PyBombs.Packager.apt-get - DEBUG - Package cmake has version 3.2.2 in dpkg
PyBombs.Packager.apt-get - DEBUG - Package cmake-data has version 3.2.2 in dpkg
PyBombs.install - DEBUG - Install tree:
|
\- rtl-sdr
PyBombs.PackageManager - DEBUG - Checking if package rtl-sdr is installed.
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.install - INFO - Installing package: rtl-sdr
PyBombs.PackageManager - DEBUG - install(rtl-sdr, static=False)
PyBombs.PackageManager - DEBUG - Package rtl-sdr is requesting a source build.
PyBombs.PackageManager - DEBUG - Using packager source
PyBombs.Packager.source - DEBUG - State on package rtl-sdr is 10
PyBombs.Packager.source - DEBUG - Package rtl-sdr is already fetched.
PyBombs.Packager.source - DEBUG - Using build directory: /home/gareth/gnuradio/src/rtl-sdr/build
PyBombs.Packager.source - ERROR - Problem occurred while building package rtl-sdr:
There should be a source dir in /home/gareth/gnuradio/src/rtl-sdr, but there isn't.
PyBombs.install - ERROR - Error installing package rtl-sdr. Aborting.
```

mbr0wn, 2016-03-11:
Maybe your prefix got into a bad state -- try running

    pybombs inv rtl-sdr none

gareth8118, 2016-03-11:
Still fails:

```
$ pybombs -p gnuradio inv rtl-sdr none
Setting package state:
rtl-sdr:	none
$ pybombs -p gnuradio install rtl-sdr
|
\- rtl-sdr
PyBombs.install - INFO - Installing package: rtl-sdr
Cloning:     (100%) [=========================================================]
PyBombs.Packager.source - ERROR - Problem occurred while building package rtl-sdr:
Unable to fetch recipe rtl-sdr
PyBombs.install - ERROR - Error installing package rtl-sdr. Aborting.
```

gareth8118, 2016-03-11:
See **Update 2** above.

closed 2016-03-11 by gareth8118

mbr0wn, 2016-03-11:
There is actually a bug here -- if the fetch fails, the package should not be flagged 'fetched'.

gareth8118, 2016-03-11:
Is there anything I can do to help reproduce and narrow the issue?

mbr0wn, 2016-03-11:
If you can come up with a way to reproduce this easily, a fix is probably very simple. Thanks!

gareth8118, 2016-03-12:
I can reproduce at work on Monday by turning the SOCKS handler off again;
I'll have a poke around in the code to see why it's not reporting properly
once I've done that.