gnuradio/pybombs #279: missing:  GNURADIO_RUNTIME_LIBRARIES for gr-zmqblocks

joek546, 2016-03-12:
This is the error I get when I try to install gr-zmqblocks

```
PyBombs.Packager.source - DEBUG - Configuring recipe gr-zmqblocks
PyBombs.Packager.source - DEBUG - Using vars - {'config_opt': '', 'builddir': '/home/sdrl-vm/target/src/gr-zmqblocks/build'}
PyBombs.Packager.source - DEBUG - In cwd - /home/sdrl-vm/target/src/gr-zmqblocks/build
PyBombs._process_thread() - DEBUG - Executing command `CC= CXX= cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/home/sdrl-vm/target  -Wno-dev'
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- checking for module 'gnuradio-runtime'
--   package 'gnuradio-runtime' not found
-- Could NOT find GNURADIO_RUNTIME (missing:  GNURADIO_RUNTIME_LIBRARIES) 
CMake Error at CMakeLists.txt:102 (message):
  GnuRadio Runtime required to compile zmqblocks

-- Configuring incomplete, errors occurred!
See also "/home/sdrl-vm/target/src/gr-zmqblocks/build/CMakeFiles/CMakeOutput.log".
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 1
PyBombs.Packager.source - ERROR - Problem occurred while building package gr-zmqblocks:
Process returned value: 1
PyBombs.install - ERROR - Error installing package gr-zmqblocks. Aborting.

```

Is gnuradio-runtime something I need to install separately?  Is there a recipe which installs this?

mbr0wn, 2016-03-12:
It's part of GNU Radio. This might be related to the other issue (regarding the envs being ignored).

mbr0wn, 2016-03-12:
Can you try latest master? It may fix this.

mbr0wn, 2016-03-14:
Closing for now--I can't reproduce, and issue #273 would explain this behaviour (and its fixed).

closed 2016-03-14 by mbr0wn