gnuradio/pybombs #281: Install crashes at iso8601 (0.1.10)

RokCam, 2016-03-22:
Trying to install on Ubuntu 15.04 (Vivid) and crash at iso8601. As per the Plex issue i tried 
> pip install iso8601
that did not help, any suggestions ?

------------------------------------------------------------
/usr/local/bin/pip run on Tue Mar 22 19:07:27 2016
adium-theme-ubuntu (0.3.4)
alabaster (0.7.7)
apt-xapian-index (0.46)
argparse (1.2.1)
astroid (1.2.1)
Babel (2.2.0)
backports-abc (0.4)
backports.ssl-match-hostname (3.5.0.1)
BeautifulSoup (3.2.1)
beautifulsoup4 (4.3.2)
binwalk (2.0.0)
bitarray (0.8.1)
Brlapi (0.6.2)
bzr (2.7.0.dev1)
ccsm (0.9.12.1)
CDApplet (1.0)
CDBashApplet (1.0)
certifi (2015.11.20.1)
characteristic (0.1.0)
chardet (2.3.0)
Cheetah (2.4.4)
colorama (0.3.2)
compizconfig-python (0.9.12.1)
configglue (1.1.2)
configobj (5.0.6)
crcmod (1.7)
cvxopt (1.1.4)
dblatex (0.3.5-2)
debtagshw (0.1)
decorator (3.4.0)
defer (1.0.6)
dirspec (13.10)
dnspython (1.12.0)
docutils (0.12)
doxyqml (0.2.0)
duplicity (0.7.1)
ecdsa (0.11)
ephem (3.7.6.0)
functools32 (3.2.3.post2)
globs (0.2)
gnome-app-install (0.4.7+nmu2ubuntu1)
gnuplot-py (1.8)
gottengeography (2.2)
gps (3.11)
guidata (1.6.1)
guiqwt (2.3.1)
h5py (2.2.1)
html5lib (0.999)
httplib2 (0.9)
idna (0.9)
IMDbPY (5.0)
impacket (0.9.12)
iotop (0.6)
ipykernel (4.2.2)
ipyparallel (4.1.0)
ipython (2.3.0)
ipython-genutils (0.1.0)
iso8601 (0.1.10)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/list.py", line 80, in run
    self.run_listing(options)
  File "/usr/lib/python2.7/dist-packages/pip/commands/list.py", line 142, in run_listing
    self.output_package_listing(installed_packages)
  File "/usr/lib/python2.7/dist-packages/pip/commands/list.py", line 151, in output_package_listing
    if dist_is_editable(dist):
  File "/usr/lib/python2.7/dist-packages/pip/util.py", line 367, in dist_is_editable
    req = FrozenRequirement.from_dist(dist, [])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 299, in from_dist
    assert len(specs) == 1 and specs[0][0] == '=='
AssertionError

iharsh234, 2016-03-22:
Try this: 
pip install -U distribute (which upgrade setuptool which causing this and it happens only with pip 1.4)
something similiar happens with me and this sort out the problem for me.

mbr0wn, 2016-03-22:
Yeah this is a pip issue.

closed 2016-03-22 by mbr0wn

RokCam, 2016-03-23:
Thanks Vijay,

did not work unfortunately

Cam

RokCam, 2016-03-26:
For anyone else having Pip issues with Ubuntu (15.04), this is how i 
fixed my problems

https://bugs.launchpad.net/ubuntu/+source/python-pip/+bug/1306991

sudo apt-get remove python-pip
wget https://raw.github.com/pypa/pip/master/contrib/get-pip.py 
<https://raw.github.com/pypa/pip/master/contrib/get-pip.py>
sudo python get-pip.py

had to modify a bit but eventually it worked.

wget https://raw.github.com/pypa/pip/master/contrib/get-pip.py
python get-pip.py
python get-https://bootstrap.pypa.io/get-pip.py
wget https://bootstrap.pypa.io/get-pip.py
python get-pip.py
python get-pip.py.1