gnuradio/pybombs #282: Gnuradio install fails: Package does not exist: python

noc0lour, 2016-03-29:
I'm trying to setup a prefix with a fresh gnuradio and some OOT-Modules. But after installing pybombs from repository and doing a 'prefix init' I'm stuck with this dependency error. 
PyBOMBS is not going to install gnuradio because of 'missing' the dependency python. I even installed it seperatly with 'pybombs install python' but installing gnuradio/gr-osmosdr still fails.

Error message is: 'PyBombs.DepManager - ERROR - Package does not exist: python (declared as dependency for package mako)'

I don't know at which point there could be an error. Maybe someone else has an idea?

mbr0wn, 2016-03-31:
Never seen this before. Can you update your gr-recipes (I fixed some minor things) and post the output of

    $ pybombs -v recipes list -l "^python$"

? (Even if it works)

Also, what's your system?

noc0lour, 2016-03-31:
After updating gr-recipes and running your command I get this output:
```
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/andrej/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'default', 'cmakebuildtype': 'RelWithDebInfo', 'cc': '/usr/lib/ccache/bin/cc', 'makewidth': 4, 'packagers': 'pip,cmd', 'gituser': 'noc0lour', 'satisfy_order': 'src', 'cxx': '/usr/lib/ccache/bin/c++', 'prefix': '/home/andrej/Projekte/pybombs', 'builddocs': False, 'timeout': 30}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/andrej/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias default.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/andrej/Projekte/pybombs/default)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/andrej/Projekte/pybombs/default/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/andrej/Projekte/pybombs/default/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/andrej/Projekte/pybombs/default/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/andrej/Projekte/pybombs/default/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/andrej/Projekte/pybombs/default/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/andrej/Projekte/pybombs/default/.pybombs/recipes', '/home/andrej/.pybombs/recipes/gr-etcetera', '/home/andrej/.pybombs/recipes/gr-recipes', '/usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/Projekte/pybombs/default/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/Projekte/pybombs/default/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 6 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.recipes - DEBUG - Initializing command class for command recipes
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['source']
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/Projekte/pybombs/default/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/Projekte/pybombs/default/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 6 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
[... cut ...]
Loading package information....PyBombs.PackageManager - DEBUG - Checking if package python is installed.

Package Name  Recipe Filename                           Installed By  
----------------------------------------------------------------------
python        ~/.pybombs/recipes/gr-recipes/python.lwr  source 
```

Currently I am running a Gentoo Linux with Kernel 4.4.6 and Python 2.7.10/Python 3.4.3 installed.
I could take a closer look and find out where the problem is but I have no clue where it might hide.

mbr0wn, 2016-03-31:
Looks like your previous call to `pybombs install python` borked your inventory. Try

    $ pybombs [-p foo] inv python none

to reset and then post again please.

noc0lour, 2016-04-03:
I have reset python the way you suggested but `pybombs install gnuradio` still fails und this is the output of `pybombs -v recipes list -l "^python$"`:
```
PyBombs.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {}
PyBombs.ConfigManager - DEBUG - Reading config info from file: /home/andrej/.pybombs/config.yml
PyBombs.ConfigManager - DEBUG - New config items: {'default_prefix': 'default', 'cmakebuildtype': 'RelWithDebInfo', 'cc': '/usr/lib/ccache/bin/cc', 'makewidth': 4, 'packagers': 'pip,cmd', 'gituser': 'noc0lour', 'satisfy_order': 'src', 'cxx': '/usr/lib/ccache/bin/c++', 'prefix': '/home/andrej/Projekte/pybombs', 'builddocs': False, 'timeout': 30}
PyBombs.ConfigManager - DEBUG - Template directory: /usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/templates
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /etc/pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/andrej/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Resolving prefix alias default.
PyBombs.ConfigManager.PrefixInfo - DEBUG - Using default_prefix as prefix (/home/andrej/Projekte/pybombs/default)
PyBombs.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/andrej/Projekte/pybombs/default/.pybombs
PyBombs.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/andrej/Projekte/pybombs/default/.pybombs/config.yml
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/andrej/Projekte/pybombs/default/src
PyBombs.Inventory - DEBUG - Trying to load inventory file /home/andrej/Projekte/pybombs/default/.pybombs/inventory.yml...
PyBombs.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/andrej/Projekte/pybombs/default/.pybombs/recipes
PyBombs.ConfigManager - DEBUG - Full list of recipe locations: ['/home/andrej/Projekte/pybombs/default/.pybombs/recipes', '/home/andrej/.pybombs/recipes/gr-etcetera', '/home/andrej/.pybombs/recipes/gr-recipes', '/usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes']
PyBombs.ConfigManager - DEBUG - Named recipe locations: {'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/Projekte/pybombs/default/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/Projekte/pybombs/default/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 6 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.recipes - DEBUG - Initializing command class for command recipes
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['source']
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/Projekte/pybombs/default/.pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/Projekte/pybombs/default/.pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 0 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-etcetera
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-etcetera' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 51 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /home/andrej/.pybombs/recipes/gr-recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/home/andrej/.pybombs/recipes/gr-recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 158 new recipes.
PyBombs.RecipeListManager - DEBUG - Loading templates.
PyBombs.RecipeListManager - DEBUG - Adding recipe location: /usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes
PyBombs.RecipeListManager - DEBUG - Scanning directory '/usr/lib64/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipes' for recipes...
PyBombs.RecipeListManager - DEBUG - Found 6 new recipes.
[... cut ...]
  packagers: pip,cmd
  prefix: /home/andrej/Projekte/pybombs
  satisfy_order: src
  timeout: 30
prefix_aliases:
  default: /home/andrej/Projekte/pybombs/default
recipes:
  gr-etcetera: git+https://github.com/gnuradio/gr-etcetera.git
  gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
packages:
    gnuradio:
        forcebuild: True
        forceinstalled: False
categories:
    common:
        forcebuild: True
```

My PyBOMBS is installed from current master branch with `setup.py build + install`

mbr0wn, 2016-04-04:
```{.yml}
config:
  packagers: pip,cmd
```

What's this about? You are disabling other packagers. Also, can you confirm that your `python.lwr` contains this:

```{.yml}
category: baseline
inherit: autoconf
satisfy:
  deb: python2.7 && python-dev
  rpm: python-devel >= 2.7
  cmd: python --version >= 2.7
vars:
  config_opt: --enable-shared
```

...otherwise you have an old recipe.

noc0lour, 2016-04-04:
I have removed `satisy_order = src` from my config and now I have some other, reasonable error: `Package does not exist: ssl (declared as dependency for package libevent)`.  The reason for this new error should be, that there is no satisfy option for something other than deb and rpm.
And I have neither of them available. I could figure out what check is missing for other installation methods (e.g. from source) and add them.

mbr0wn, 2016-04-04:
Good bet is probably to add `pkg-config` option. However, seems like we can close this.

closed 2016-04-04 by mbr0wn

stuzog, 2017-01-01:
I'm having the same error come up with trying to install gnuradio on OS X 10.11:

```
PyBOMBS.DepManager - ERROR - Package has no install method: ssl (declared as dependency for package lib event)
```
I ran the recipe list as above and the last lines were:

```
Package Name  Recipe Filename                           Installed By  
----------------------------------------------------------------------
python        ~/.pybombs/recipes/gr-recipes/python.lwr  cmd     
```

Any suggestions?