gnuradio/pybombs #285: Syntax error in config leads to incorrect error

ckuethe, 2016-04-05:
If I use tabs in my config.yml that leads to an error of the form "Package XYZ has no recipe file!"

Possibly the yml parser could be made more tolerant of whitespace errors, or a more clear error message should be printed: "syntax error in $configfile, check whitespace"

ckuethe, 2016-04-11:
Proposed fix: #295

ckuethe, 2016-04-12:
Assuming no one has any objections, I plan to merge my patch tomorrow afternoon:  well-formed config files still parse, malformed ones now cause useful error messages to be printed.

ckuethe, 2016-04-12:
Fixed.

closed 2016-04-12 by ckuethe

mbr0wn, 2016-04-12:
Good fix, thanks. But please stick with Python3-compatible code and prefer .format() over %.

mbr0wn, 2016-04-12:
Also, safe_load can return 'None' without an error. This was formerly caught by the `or {}`. Will need to add that back.

ckuethe, 2016-04-12:
ACK.