gnuradio/pybombs #286: Issue with "mako" recipe

mhostetter, 2016-04-08:
I just tried to install GNU Radio with PyBOMBS on a fresh Ubuntu 14.04 image.  There was a python error when running `$ pybombs install gnuradio`.  The error is shown below.  I started installing some of the dependencies independently and was successful.  I traced the error back to an issue when installing mako.  When running `$ pybombs install mako`, I get this (same) error:

`matt@t430:~/pybombs$ pybombs install mako
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.0.1', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/main.py", line 30, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/base.py", line 141, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/install.py", line 129, in run
    self._check_if_pkg_goes_into_tree if not self.args.no_deps else lambda x: bool(x in self.args.packages)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/dep_manager.py", line 56, in make_dep_tree
    dep_trees[pkg] = self.make_tree_recursive(pkg, filter_callback)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/dep_manager.py", line 76, in make_tree_recursive
    if not self.pm.exists(pkg):
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/package_manager.py", line 113, in exists
    pkg_version = pkgr.exists(r)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/packagers/extern.py", line 102, in exists
    return self._packager_run_tree(recipe, self._package_exists)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/packagers/extern.py", line 163, in _packager_run_tree
    return satisfy_rule.ev(satisfy_evaluator)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipe.py", line 53, in ev
    return func(self.name, self.compare or ">=", self.version)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/packagers/extern.py", line 172, in _package_exists
    or (required_version is not None and not vcompare(comparator, available_version, required_version)):
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/vcompare.py", line 34, in vcompare
    return operators[cmp_op](LooseVersion(version_x), LooseVersion(version_y))
  File "/usr/lib/python2.7/distutils/version.py", line 265, in __init__
    self.parse(vstring)
  File "/usr/lib/python2.7/distutils/version.py", line 274, in parse
    self.component_re.split(vstring))
TypeError: expected string or buffer`

I did a `$ sudo apt-get install python-mako` and then ran `$ pybombs install gnuradio` and was successful.

mbr0wn, 2016-04-08:
Can you unscrew the stack trace please? Also which version of PyBOMBS?

mhostetter, 2016-04-08:
I just pulled master, so 9d21f27.

If by unscrew you mean make the stack trace easier to read, then yes.

```
matt@t430:~/pybombs$ pybombs install mako
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.0.1', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/main.py", line 30, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/base.py", line 141, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/install.py", line 129, in run
    self._check_if_pkg_goes_into_tree if not self.args.no_deps else lambda x: bool(x in self.args.packages)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/dep_manager.py", line 56, in make_dep_tree
    dep_trees[pkg] = self.make_tree_recursive(pkg, filter_callback)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/dep_manager.py", line 76, in make_tree_recursive
    if not self.pm.exists(pkg):
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/package_manager.py", line 113, in exists
    pkg_version = pkgr.exists(r)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/packagers/extern.py", line 102, in exists
    return self._packager_run_tree(recipe, self._package_exists)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/packagers/extern.py", line 163, in _packager_run_tree
    return satisfy_rule.ev(satisfy_evaluator)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/recipe.py", line 53, in ev
    return func(self.name, self.compare or ">=", self.version)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/packagers/extern.py", line 172, in _package_exists
    or (required_version is not None and not vcompare(comparator, available_version, required_version)):
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/utils/vcompare.py", line 34, in vcompare
    return operators[cmp_op](LooseVersion(version_x), LooseVersion(version_y))
  File "/usr/lib/python2.7/distutils/version.py", line 265, in __init__
    self.parse(vstring)
  File "/usr/lib/python2.7/distutils/version.py", line 274, in parse
    self.component_re.split(vstring))
TypeError: expected string or buffer
```

mbr0wn, 2016-04-08:
Thanks, much better. Doesn't immediately clue me in though... Lemme see what I can find out.

mbr0wn, 2016-04-13:
OK, @namccart found the problem and has fixed it on the recipes side. I'll see if I can fix this in here too.

mhostetter, 2016-04-13:
Sweet, thanks guys!

mbr0wn, 2016-04-17:
OK, submitted a fix so this won't break so easily in the future. Thanks for bringing this up.

closed 2016-04-17 by mbr0wn

mhostetter, 2016-04-17:
Thanks for the help!