gnuradio/pybombs #288: Make smaller gnuradio installations?

stwunsch, 2016-04-08:
Hi,

Is there a possibility to make smaller gnuradio installations? The current recipe just installs everything. It would be nice if it could be possible to do a kind of 'gnuradio base' installations.

I thought about a second gnuradio recipe, which installs just the most necessary things. But this solution would be little bit messy.

Or is it possible to manipulate the config_opt variable from the user interface? So you could disable some components via the ENABLE_STUFF=False flags. Sth like 'pybombs install --flag "-DENABLE_WX=False" gnuradio'.

Greetings
Stefan

ckuethe, 2016-04-08:
I'll check to see what flags can be passed through the environment. It might be interesting to pass different flags for each recipe, since I'm not sure that you'd necessarily want to pass some flags to all builds:

    env PB_gnuradio_CFLAGS="this" PB_librtsdr_CFLAGS="that" PB_gqrx_CFLAGS=other pybombs install gqrx

Are you trying to optimize for build time or install size?

stwunsch, 2016-04-08:
I am trying to optimize for build time because I want to compile it for docker using pybombs, but I have a 2h build time limitation (on a single core).

ckuethe, 2016-04-08:
I get why you're asking; I never bother to build the documentation on my machine.

I'm unconvinced that skipping certain parts of gnuradio would be very useful. It seems like you'd inevitably find yourself missing some module, possibly as soon as you tried to link some other app.

To make this be really useful we'd probably have to make the gnuradio build produce a bunch of artifacts that other recipes would then depend on.

ckuethe, 2016-04-08:
That said, I'll still take a look at being able to pass in some environment variables.

mbr0wn, 2016-04-08:
You can override pretty much anything inside a recipe in your local .pybombs/config.yml. That includes the config_opt vars, and the config_filter.

mbr0wn, 2016-04-08:
I'd say this feature request is halfway between already implemented (see my previous comment) and #292. Closing.

closed 2016-04-08 by mbr0wn