gnuradio/pybombs #290: Using python-requests on ftp URLs fails

marcusmueller, 2016-04-08:
… because requests is an http library. There is a requests-ftp adapter, but I don't think we should be using that. 

There's different possible solutions:

1. Simply find http mirror URLs or git sources for the *three* known recipes that use ftp
 * that might call for the `git.py` fetcher to be aware of tags so PyBOMBS can fetch releases (that'd be a neat feature, anyway); not a great problem, but still.
 * In the long run: deprecate the `wget+http://` schemes; use `http://` (or create an alias); it's not like `requests` is anything but a HTTP handler
2. simply workaround that using urllib for ftp URLs and live without the download progress bar
3. install, and use, `requests-ftp` and let it hotpatch `requests`.
 * Kind of fragile, because no-one knows when that will stop working

ckuethe, 2016-04-08:
I'll try track down some http urls.

marcusmueller, 2016-04-08:
Thanks, the problem is especially Alsa; it has plenty of mirrors, but
some aren't really trustworthy or haven't been updated since 1.0.22
(current recipe has 1.0.27, current release is 1.1.1); fedora has a copy
of that on their servers,
http://pkgs.fedoraproject.org/repo/pkgs/alsa-lib/alsa-lib-1.1.1.tar.bz2/881060d2c568f7f49db82c58df2f9ddc/alsa-lib-1.1.1.tar.bz2
but I don't know how long that URL is gonna be around.
I think switching to git for alsa might be worth it.

ckuethe, 2016-04-08:
https://github.com/gnuradio/gr-recipes/commit/487bdaed1c502b99fced3ec9847c827796722dd4

I went with the tux.org mirror.

marcusmueller, 2016-04-08:
I'll close this issue then. Unless someone uses an FTP URL in the future, things are solved.

On the other hand, I'm still partial to replacing `wget+http://` with `http://`. What's your opinion on adding that alias?

closed 2016-04-08 by marcusmueller

ckuethe, 2016-04-08:
Assume an implicit "wget" prefix unless you specify a different fetcher
(like git+ or svn+ or hg+)? Favorable.

marcusmueller, 2016-04-08:
Uh, wait, now I feel stupid; isn't that why
https://github.com/gnuradio/pybombs/blob/master/pybombs/fetchers/base.py
defines `regexes[]`?

Now I'll have to try :)

marcusmueller, 2016-04-08:
yeah. works.

ckuethe, 2016-04-08:
Just be careful of github urls and doing the right thing with a repo name (with and without the .git suffix) and a download blob. Not that any recipies are using it right now, but it's worth considering.

marcusmueller, 2016-04-08:
I just removed all the `wget+`; can't really hurt github URLs that way.

mbr0wn, 2016-04-08:
DON'T REMOVE THE wget+