gnuradio/pybombs #294: add 'builddir' variable, stop abusing 'installdir'

ckuethe, 2016-04-09:
While fixing https://github.com/gnuradio/gr-etcetera/issues/3 I found that the build dir is set from the 'installdir' variable (https://github.com/gnuradio/pybombs/blob/master/pybombs/packagers/source.py#L212). Kind of confusing. Perhaps there should be a 'builddir' variable containing 'build' unless otherwise overridden.

mbr0wn, 2016-04-12:
I agree -- the current behaviour was inherited from PB1. I hate to just break stuff though. Do you have any suggestions that don't break any recipes? Maybe use both?

ckuethe, 2016-04-12:
I'll do up a patch to prefer 'builddir' and warn when 'installdir' is used, then we can push that to pip (which will also pick up the updated build-essential change.

Once that's available I'll update the recipes.

ckuethe, 2016-04-13:
Fixed by #299

mbr0wn, 2016-05-02:
#299 actually broke a bunch of things, so it was reverted and this bug is thus open again.