gnuradio/pybombs #297: something about setup_env.sh

ckuethe, 2016-04-12:
Legacy pybombs used to have the "env" command which would generate some environmental setup shell statements.

- we should probably bring that back for ... unusual ... installation prefixes
- we should mention important path variables in the quickstart for people installing into strange places.

See https://github.com/mossmann/hackrf/issues/248 for more context.

mbr0wn, 2016-04-12:
`prefix init` does just that.

ckuethe, 2016-04-12:
I can't run "pybombs prefix init" on an existing prefix. Would it be worth adding an "env" subcommand to rewrite the setup_env.sh file?

mbr0wn, 2016-04-12:
No, but it would be worth fixing `prefix init` to work on an existing prefix.

mbr0wn, 2016-04-12:
...or add a command line switch just to write that file for the `prefix` subcommand.

ckuethe, 2016-04-12:
That's what I meant by subcommand. Something like  "pybombs prefix env" ... which I see is already in use, so "prefix install-env" maybe?

mbr0wn, 2016-04-13:
OK, that works for me. Do you want to add this?

ckuethe, 2016-04-13:
I can do it

ckuethe, 2016-04-13:
PR #300

closed 2016-04-13 by ckuethe