gnuradio/pybombs #298: improve documentation of subcommands

ckuethe, 2016-04-12:
"pybombs prefix env --help" (for example) doesn't tell me what the command does, only what arguments it could accept.

ckuethe, 2016-04-13:
I'm thinking that "pybombs help XYZ" and "pybombs XYZ help" should do the same (useful) thing...

    $ pybombs help prefix
    usage: pybombs prefix [-h] {info,init,install-sdk,env} ...
    
    positional arguments:
      {info,init,install-sdk,env}
                            Prefix Commands:
        info                Display information on the currently used prefix.
        init                Create and initialize a prefix.
        install-sdk         Install an SDK into the prefix.
        env                 Print the environment variables used in this prefix.
    
    optional arguments:
      -h, --help            show this help message and exit
    $ pybombs prefix help
    usage: pybombs prefix [-h] {info,init,install-sdk,env} ...
    pybombs prefix: error: argument prefix_command: invalid choice: 'help' (choose from 'info', 'init', 'install-sdk', 'env')

mbr0wn, 2016-12-10:
Fixed `pybombs prefix env --help` (and all others like this). Putting help everywhere is tricky, and not worth the effort IMO. It's probably easier to rip it out entirely.

closed 2016-12-10 by mbr0wn