gnuradio/pybombs #301: Newb questions about Pybombs 2

dsound, 2016-04-17:
Having been away for a while (2 years) and having tried Pybombs 1, I was understandable surprised to see the latest development with Pybombs 2. I'm using latest distro of Kali and can't seem to get Pybombs working. I've used the suggested method of downloading using PIP. It installs Pybombs in /usr/bin/local but when I try to run any of the suggested scripts, they obviously won't work because there is no 'setup.py' When I try to run 'pybombs install gnu-radio' I get an error that there is no recipe for gnuradio. When trying to run 'pybombs --prefix-conf PREFIX_CONF' I get the following output:

  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.0.1', 'console_scripts', 'pybombs')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 568, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2720, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2380, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2386, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/main.py", line 25, in <module>
    from pybombs.commands import dispatch
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/__init__.py", line 23, in <module>
    from base import CommandBase, dispatch
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/base.py", line 25, in <module>
    from pybombs.config_manager import config_manager
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/config_manager.py", line 636, in <module>
    config_manager = ConfigManager()
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/config_manager.py", line 315, in __init__
    self.load(select_prefix)
  File "/usr/local/lib/python2.7/dist-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/config_manager.py", line 325, in load
    args = parser.parse_known_args()[0]
  File "/usr/lib/python2.7/argparse.py", line 1733, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python2.7/argparse.py", line 1939, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python2.7/argparse.py", line 1879, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python2.7/argparse.py", line 1791, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python2.7/argparse.py", line 2231, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python2.7/argparse.py", line 2260, in _get_value
    result = type_func(arg_string)
IOError: [Errno 2] No such file or directory: 'PREFIX_CONF'

mbr0wn, 2016-04-17:
Please ask such questions on the mailing list, this issue tracker is for bug reports. In your case, make sure you have the right package name (e.g. gnuradio instead of gnu-radio) and I'm not sure what you were trying to do with the second command.

closed 2016-04-17 by mbr0wn

dsound, 2016-04-17:
Ok, sorry about that. I just don't know what I'm doing wrong here. I can't
figure out how to access the recipes to install gnuradio and others.