gnuradio/pybombs #302: pybombs / recipes / uhd.lwr @ 14f73067 is not setting bulddir

HTlocal, 2016-04-17:
Hi,

on my system I had to set
~/.pybombs/recipes/gr-recipes/uhd.lwr
+ builddir: host/build
to get it working.

Cheers,
local

gr-sp, 2016-04-20:
I have been having similar issues with many other recipes. I noticed that recently the templates were updated to contain a "default" value for builddir.

What happens if my recipe defines a value for installdir? The file packagers/source.py checks for recipe.builddir first and if it doesn't exist it uses the value of installdir. If the template has a builddir defined, wouldn't the test in source.py always be true?

If the answer to this question is "yes", then that would explain everything I have been experiencing lately. If the answer is "no" then I am not sure what is going on here.

mbr0wn, 2016-04-21:
I'm reverting the commit that broke this for now -- we'll get back to the whole installdir thing later.

closed 2016-04-21 by mbr0wn