gnuradio/pybombs #306: prefix init fails

gr-sp, 2016-04-22:
> Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 9, in <module>
    load_entry_point('PyBOMBS==2.0.1', 'console_scripts', 'pybombs')()
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/main.py", line 29, in main
    return dispatch() or 0
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/base.py", line 141, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/prefix.py", line 102, in run
    return self.prefix_cmd_name_list[self.args.prefix_command]['run'](self)
  File "/usr/local/lib/python2.7/site-packages/PyBOMBS-2.0.1-py2.7.egg/pybombs/commands/prefix.py", line 152, in _run_init
    if op.exists(op.join(path, self.prefix.prefix_conf_dir)):
AttributeError: 'NoneType' object has no attribute 'prefix_conf_dir'

I just cloned a fresh copy of pybombs and went to init my prefix and got the above Traceback. I took a quick peek at the code. This occurs in prefix.py during a test to determine if the prefix folder contains an existing prefix. Obviously from the above Trackback the self.prefx object doesn't appear to contain the prefix_conf_dir attribute (which is the ".pybombs" subfolder of the prefix from what I can gather from the code).

gr-sp, 2016-04-22:
In my local copy, I basically rolled back commit [prefix: Minor cleanup in prefix init](https://github.com/gnuradio/pybombs/commit/93ea53f54525dea89890282fcb79f3f7a0772aa6) and that seems to have worked.

mbr0wn, 2016-04-24:
Yeah just submitted a hotfix. Not a permanent solution...

closed 2016-04-24 by mbr0wn