gnuradio/pybombs #307: qwt5 source build "reprefix"

estatz, 2016-04-25:
Old pybombs created an environment variable, "reprefix", that was an escaped version of the "prefix" string that was safe to pass in to sed.  [Link](https://github.com/gnuradio/pybombs_legacy/blob/master/mod_pybombs/vars.py#L46)

Pybombs2 does not appear to provide "reprefix", but "reprefix" is still used when building qwt5 from source.  [Link](https://github.com/gnuradio/gr-recipes/blob/master/qwt5.lwr#L21)

mbr0wn, 2016-04-26:
Good catch! I'm not a fan of hacking PyBOMBS just to suit individual recipes. The qwt5 recipe could do the escaping, something like `python -c "import os; import re; print re.escape(os.environ['PYBOMBS_PREFIX])"`. This way we don't have to pollute the core code just to satisfy the QWT recipe, when that should be the one eating its own complexity.

Can you please provide a PR for this recipe?

mbr0wn, 2016-04-26:
Transferring to: https://github.com/gnuradio/gr-recipes/issues/20

closed 2016-04-26 by mbr0wn