gnuradio/pybombs #308: Pybombs 2.0.1 install fails with pip list error

jonbram, 2016-04-25:
So I'm new to gnuradio so I have been following the pybombs install instructions. PyBombs itself seems to install ok and I have set up recipes. When I try an install a recipe, I get an error

$pybombs install gnuradio
PyBombs.Packager.pip - ERROR - Could not run pip list. Hm.
PyBombs.Packager.pip - ERROR - Command '['pip', 'list']' returned non-zero exit status 2

I attached a verbose version of the same command.

I also found that I get the same error when I try and list the recipes

$ pybombs recipes list
Loading package information................................................................................................................................................................PyBombs.Packager.pip - ERROR - Could not run pip list. Hm.
PyBombs.Packager.pip - ERROR - Command '['pip', 'list']' returned non-zero exit status 2

Any help would be much appreciated.

Jonathan

[pybombs_error.txt](https://github.com/gnuradio/pybombs/files/235611/pybombs_error.txt

ckuethe, 2016-04-25:
"pip list" failed. Try that in a terminal and see why. Perhaps you don't
have python-pip installed?

mbr0wn, 2016-04-26:
...or you have a broken pip installed. Ubuntu 14.04 ships a broken pip. If that's the case, you can update pip with easy_install.

jonbram, 2016-04-26:
That seemed to be it. But I had to uninstall pip first and delete the 
dist-package before i could get it to re install with easy_install.

Once this was done pybombs at least started to work, but this showed 
that boost was failing to install which sent me down a path of 
installing libboot1.54-dev and everything it depended on.

gnuradio is now building but it is now failing building UHD with

PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 0
PyBombs.Packager.source - DEBUG - Configure successful.
PyBombs.Inventory - DEBUG - Setting state to `configured'
PyBombs.Inventory - DEBUG - Saving inventory to file 
/home/jonathan/gnuradio/.pybombs/inventory.yml...
PyBombs.Packager.source - DEBUG - Building recipe uhd
PyBombs.Packager.source - DEBUG - In cwd - 
/home/jonathan/gnuradio/src/uhd/host/build
PyBombs._process_thread() - DEBUG - Executing command `make -j4'
|
+- liblog4cpp
|
\- uhd
|
\- gnuradio
    |
    +- liblog4cpp
    |
    \- uhd
Cloning:     (100%) 
[=========================================================]
Configuring: (100%) 
[=========================================================]
Building:    ( 76%) [===========================================/ 
]PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 2
PyBombs.Packager.source - ERROR - Problem occurred while building 
package uhd:
Process returned value: 2
PyBombs.install - ERROR - Error installing package uhd. Aborting.
Building:    (100%) 
[=========================================================]

Thanks for the help

Jonathan

mbr0wn, 2016-04-26:
OK, closing this issue then. Please ask support questions on discuss-gnuradio. You may have an older version of PyBOMBS btw.

mbr0wn, 2016-04-26:
...if this is an issue with the current pybombs codebase, please open another issue.

closed 2016-04-26 by mbr0wn