gnuradio/pybombs #311: Cheetah NameMapper C Version

mhostetter, 2016-04-29:
On a fresh image of Ubuntu 14.04 LTS, I installed GNU Radio using the most recent PyBOMBS (installed through pip).  After the install I get this error when trying to run `gnuradio-companion`.

```
matt@series5:~/repos/gr-adsb/examples$ gnuradio-companion
/usr/local/lib/python2.7/dist-packages/Cheetah/Compiler.py:1509: UserWarning: 
You don't have the C version of NameMapper installed! I'm disabling Cheetah's useStackFrames option as it is painfully slow with the Python version of NameMapper. You should get a copy of Cheetah with the compiled C version of NameMapper.
  "\nYou don't have the C version of NameMapper installed! "
/usr/local/lib/python2.7/dist-packages/Cheetah/Compiler.py:1509: UserWarning: 
You don't have the C version of NameMapper installed! I'm disabling Cheetah's useStackFrames option as it is painfully slow with the Python version of NameMapper. You should get a copy of Cheetah with the compiled C version of NameMapper.
  "\nYou don't have the C version of NameMapper installed! "
/usr/local/lib/python2.7/dist-packages/Cheetah/Compiler.py:1509: UserWarning: 
You don't have the C version of NameMapper installed! I'm disabling Cheetah's useStackFrames option as it is painfully slow with the Python version of NameMapper. You should get a copy of Cheetah with the compiled C version of NameMapper.
  "\nYou don't have the C version of NameMapper installed! "
<<< Welcome to GNU Radio Companion v3.7.9.2-191-gb5b93342 >>>

Block paths:
	/home/matt/pybombs/target/share/gnuradio/grc/blocks

Loading: "/home/matt/repos/gr-adsb/examples/adsb_rx.grc"
/usr/local/lib/python2.7/dist-packages/Cheetah/Compiler.py:1509: UserWarning: 
You don't have the C version of NameMapper installed! I'm disabling Cheetah's useStackFrames option as it is painfully slow with the Python version of NameMapper. You should get a copy of Cheetah with the compiled C version of NameMapper.
  "\nYou don't have the C version of NameMapper installed! "
/usr/local/lib/python2.7/dist-packages/Cheetah/Compiler.py:1509: UserWarning: 
You don't have the C version of NameMapper installed! I'm disabling Cheetah's useStackFrames option as it is painfully slow with the Python version of NameMapper. You should get a copy of Cheetah with the compiled C version of NameMapper.
  "\nYou don't have the C version of NameMapper installed! "
```

The NameMapper errors continue for several hundred lines of error messages.  Some googling seems to suggest Cheetah must be built from source to get the C-version of NameMapper.  Do you have any ideas?  Should `cheetah.lwr` be set to forcebuild?

mbr0wn, 2016-05-02:
I've never built Cheetah from source... and don't know anyone who has. Maybe this is a GRC thing? Does it work with this error/warning?

primercuervo, 2016-05-12:
This is happening in some machines, but funny enough I haven't found the exact way of replicating the issue in other machines. However, in my main computer I did get it. It was solved by getting and installing the last version of cheetah from their git repository:

https://github.com/cheetahtemplate/cheetah/tree/master

and then rebuilding gnuradio from pybombs:

pybombs rebuild gnuradio

mbr0wn, 2016-05-13:
We need to figure out how to reproduce this before we can do anything about it.

mbr0wn, 2016-05-15:
I think I know what the issue is. Cheetah through pip is different from Cheetah through apt-get. I'll remove the pip recipe and add the git source as first choice.

@mhostetter Appreciate feedback on this. You'll need to update your gr-recipes.

mhostetter, 2016-05-15:
Sorry I've been MIA on this issue.  I will update my gr-recipes and try again.  Thanks everyone for the help!

mhostetter, 2016-05-15:
At first it didn't work for me, but then I realized Cheetah was still installed on my machine from the previous installation.  After a `sudo pip uninstall cheetah` and a `pybombs install cheetah`, everything is working fine.  Great catch, @mbr0wn.

mbr0wn, 2016-05-15:
Well, that's one down :)

closed 2016-05-15 by mbr0wn