gnuradio/pybombs #315: autoconf.lwr recipe bug

drewvian, 2016-05-04:
The latest commit 496c3f03297fcf94487233c558cfadae209a90e7 broke the installation of autoconf. Running the command
`pybombs install autoconf`
or any other install requiring autoconf (all of them?) results in:
`RuntimeError: maximum recursion depth exceeded`
This follows a long wait where pybombs attempts to recursively install autoconf as a prerequisite for itself. autoconf.lwr should have line 21 removed, and then the install works properly again.

mbr0wn, 2016-05-05:
Can you pull latest gr-recipes and pybombs and see if this is still around? Should be fixed in 1e634157be3090b3b8b88bf4e12ade83e96c22ce and the corresponding gr-recipes commit.

drewvian, 2016-05-05:
That commit seems to have corrected the issue.
Thanks,
Andrew Vian

closed 2016-05-05 by mbr0wn