gnuradio/pybombs #316: Catch fetch errors when adding recipes

n-west, 2016-05-05:
If a recipe URI fails to download (for example because the ssh user does not have permission to read remote repo) a success is reported (or more accurately no message indicating something went wrong), but the recipe repo is not downloaded.

mbr0wn, 2016-05-07:
What's the easiest way to reproduce this?

mbr0wn, 2016-05-08:
I tried removing my SSH keys, and got this:

```
PyBombs.Fetcher.git - DEBUG - Requirements met.
PyBombs.Fetcher.git - DEBUG - Using url - git@github.com:EttusResearch/uhd.git
PyBombs._process_thread() - DEBUG - Executing command `['git', 'clone', 'git@github.com:EttusResearch/uhd.git', 'uhd', '-b', 'master']'
Cloning into 'uhd'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 128
PyBombs.Fetcher - DEBUG - That didn't work.
PyBombs.Fetcher - DEBUG - Process returned value: 128
PyBombs.Packager.source - ERROR - Problem occurred while building package uhd:
Unable to fetch recipe uhd
PyBombs.install_manager - ERROR - Error installing package uhd. Aborting.
```

=> what I would expect.

n-west, 2016-05-08:
I'm not using the correct language here.

```
nathan@westengineeringlabs:~$ pybombs prefix init gnuradio
PyBombs.prefix - INFO - Creating directory `/home/nathan/gnuradio'
PyBombs.prefix - INFO - Initializing PyBOMBS prefix in `/home/nathan/gnuradio'...
nathan@westengineeringlabs:~$ pybombs -p gnuradio recipes add gr-recipes git@github.com:gnuradio/gr-recipes.git
The authenticity of host 'github.com (192.30.252.121)' can't be established.
RSA key fingerprint is 16:27:ac:a5:76:28:2d:36:63:1b:56:4d:eb:df:a6:48.
Cloning:     [\                                                        yes                                                                                                                                                                    Cloning:     [ \                                                                                                                                                                                                                            ]
Cloning:     (100%) [=======================================================================================================================================================================================================================]
nathan@westengineeringlabs:~$ ls gnuradio/.pybombs/recipes/
nathan@westengineeringlabs:~$ cat gnuradio/.pybombs/config.yml 
categories:
  common:
    forcebuild: true
  hardware:
    forcebuild: true
packages:
  gnuradio:
    forcebuild: true
recipes:
  gr-recipes: git@github.com:gnuradio/gr-recipes.git
```

what's happening: I use a git URI to add a recipe repo that I don't have permission to clone. Pybombs happily shows the progress bar hitting 100% for cloning, but really nothing happens. The recipe is added to the config list.

gr-sp, 2016-05-09:
Similarly, if this is the first time that you are accessing a specific URI, SSH will ask if you want to permanently add to list of known hosts or it says something along the lines of "The authenticity of host '<host>' can't be established. RSA key fingerprint is <fingerprint>. Would you like to add to the list of know hosts?". However, pybombs immediately goes into the "cloning" progress display and obscures the prompt that is asking for input. If you respond to the hidden prompt, things work as normal but it is a little odd because you are typing your input on top of the "cloning" progress display.

mbr0wn, 2016-05-14:
@gr-sp good point, should probably be separate issue though.

mbr0wn, 2016-05-14:
e0521d7e8288cc53a9e1b5d1870a2293d55cf88b might fix this. It's a general safety measure and not tested for this issue yet.

mariobaldini, 2017-08-08:
`Unable to fetch recipe uhd`

I had this error and was due to connectivity issues, apparently. 

After changing the DNS servers to manually use OpenDNS/Google it worked. (instead of the local ISP). France telcos sometimes redirect to a warning about privacy. 

Also, if you're using some sort of proxy, some CDNs sometimes impose an captcha. 

**Using OpenDNS servers and without proxy fixed the issue.**

closed 2017-09-15 by mbr0wn