gnuradio/pybombs #324: Problem with the installation

hacene182, 2016-05-11:
Hi , I'am trying to install gnuradio using pybombs when I type " sudo python setup.py install" i get this error :
$:~/pybombs$ sudo python setup.py install 
Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-21.0.0.zip
Extracting in /tmp/tmpb7CSPl
Traceback (most recent call last):
  File "setup.py", line 71, in <module>
    ez_setup.use_setuptools()
  File "/home/hacene/pybombs/ez_setup.py", line 173, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/hacene/pybombs/ez_setup.py", line 124, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/hacene/pybombs/ez_setup.py", line 64, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/hacene/pybombs/ez_setup.py", line 103, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/hacene/pybombs/ez_setup.py", line 87, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file

NinjaComics, 2016-05-11:
@hacene182 Try installing setuptools using:

`sudo pip install setuptools`

That should solve the problem

mbr0wn, 2016-05-12:
@hacene182, do you need to install from source?

hacene182, 2016-05-12:
@mbr0wn yes .

mbr0wn, 2016-05-13:
ez_setup should make it work even without setuptools. I've had issues with ez_setup in the past, though. Maybe this is another one. @hacene182, if you run it twice, does it work?

mbr0wn, 2016-09-18:
No activity -- closing.

closed 2016-09-18 by mbr0wn