gnuradio/pybombs #326: Dependency checks take too long

gr-sp, 2016-05-11:
Dependency checks (before pybombs even begins to build anything) take on the order of 35-40 minutes on my system. I am using an older system (CentOS 6.5) such that many of the GNU Radio dependencies are not met when starting from a base system installation. The time it takes seems a little (rather a lot) excessive.

Watching the output (pybombs -v[v] install gnuradio) from the build shows the same package checks being performed repetitively and although there are statements in the output that indicate there is some caching of the state of the dependencies, the time required to perform the checks indicates otherwise.

I haven't had a chance to dive into the code for this so I don't have anything specific to point to as a problem. I am hoping that someone else knows enough about how pybombs performs dependency checking to either take a look at this or to indicate why it takes so long and whether or not anything can be done about it.

NinjaComics, 2016-05-11:
Can you post the verbose output here ?

gr-sp, 2016-05-11:
So, here is a file containing just the dependency checking part of my build with a command similar to:
`pybombs -v install gnuradio`

[dependency_checks.zip](https://github.com/gnuradio/pybombs/files/259691/dependency_checks.zip)

Just this portion of my build took on the order of 40 minutes. There are a few things to note:
- The actual build is attempting to build a GNU Radio OOT module that depends on GNU Radio and a few other OOT modules.
- There may be some statements in the output that are coming from modifications (just print statments) that I have made to my pybombs installation. Specifically the line "Creating new PackageManager" and the line that follows it containing "__init__".
- The attached file also contains the printout of the dependency tree near the end...which is enormous.
- I am using a modified set of the gr-recipes found in github/gr-recipes so the version numbers you see in the output may not be the same as typically seen in a standard GNU Radio build.
- I used 7zip to zip up the attachment.

mbr0wn, 2016-05-12:
Some persistent caching feature would be good.

mbr0wn, 2016-05-12:
FYI, the caching we currently have pertains to finding and reading the recipe.

mbr0wn, 2016-05-13:
@gr-sp I'm sure you think this is a bug, and I wouldn't disagree, but I'll classify it as enhancement for now because in a sense, PyBOMBS isn't broken, it's just too slow.

gr-sp, 2016-05-13:
Your right...it's just really annoying and time consuming.

gr-sp, 2016-05-16:
Ahh...the caching (https://github.com/gnuradio/pybombs/commit/a698320731c7021033234beba38cc5d2cdf4edcd) works nicely. We went from 30 minutes to < 1 minute for dependency checking. Thanks a ton. This will make my life much simpler.

mbr0wn, 2016-05-16:
Wow, that much? In that case, I'll just close this. Maybe we don't need a persistent thing.

closed 2016-05-16 by mbr0wn