gnuradio/pybombs #327: pybombs install pkg-config error

drewvian, 2016-05-11:
I have run into a problem with pybombs installation procedure. I just started running into this problem yesterday. Running the exact same procedure last week worked correctly. I am attempting to run the command: `pybombs install uhd gnuradio`

Simplifying that procedure down, the error occurs running the command: `pybombs install pkg-config`

I receive the following error:
```
PyBombs.install_manager - INFO - Installing package: pkg-config
Install tree:
PyBombs.Fetcher - ERROR - Unexpected error while fetching wget+https://pkgconfig.freedesktop.org/releases/pkg-config-0.28.tar.gz.
PyBombs.Fetcher - ERROR - hostname 'pkgconfig.freedesktop.org' doesn't match either of 'distributions.freedesktop.org', 'farsight.freedesktop.org', 'fontconfig.freedesktop.org', 'fontconfig.org', 'freedesktop.org', 'geoclue.freedesktop.org', 'secure.freedesktop.org', 'www.fontconfig.org', 'www.freedesktop.org'
PyBombs.Packager.source - ERROR - Problem occurred while building package pkg-config:
Unable to fetch recipe pkg-config
PyBombs.install_manager - ERROR - Error installing package pkg-config. Aborting.
```
Adding the -v flag to the command results in the following output:
```
PyBombs.Packager.apt-get - DEBUG - Package pkg-config has version 0.26 in dpkg
PyBombs.install_manager - INFO - Installing package: pkg-config
PyBombs.PackageManager - DEBUG - install(pkg-config, static=False)
PyBombs.PackageManager - DEBUG - Using packager pip
PyBombs.PackageManager - DEBUG - Using packager apt-get
PyBombs.ReqScanner - DEBUG - Parsing '(pkg-config >= 0.28)'
PyBombs.Packager.apt-get - DEBUG - Package pkg-config has version 0.26 in apt-cache
PyBombs.PackageManager - DEBUG - Using packager pkgconfig
PyBombs.PackageManager - DEBUG - Using packager cmd
PyBombs.PackageManager - DEBUG - Using packager source
PyBombs.Packager.source - DEBUG - State on package pkg-config is 0
PyBombs.Fetcher - DEBUG - Fetching source for recipe: pkg-config
PyBombs.Fetcher - DEBUG - Switching to src directory: /home/efx/gr-3.7/src
PyBombs.Fetcher - DEBUG - Getting fetcher for https://pkgconfig.freedesktop.org/releases/pkg-config-0.28.tar.gz, type wget
PyBombs.Fetcher - DEBUG - Using fetcher <pybombs.fetchers.wget.Wget object at 0x7f13259bf3d0>
PyBombs.Fetcher.wget - DEBUG - Requiring packages on host system: ['python-requests']
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - apt-get is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager port
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'apt-get', 'pkgconfig', 'cmd']
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBombs.PackageManager - DEBUG - pip is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager apt-get
PyBombs.PackageManager - DEBUG - apt-get is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager port
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBombs.PackageManager - DEBUG - pkgconfig is supported!
PyBombs.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBombs.PackageManager - DEBUG - cmd is supported!
PyBombs.PackageManager - DEBUG - Using packagers: ['pip', 'apt-get', 'pkgconfig', 'cmd']
PyBombs.PackageManager - DEBUG - Checking if package python-requests is installed.
PyBombs.ReqScanner - DEBUG - Parsing 'requests'
PyBombs.Packager.pip - DEBUG - Loading pip install cache.
PyBombs.PackageManager - DEBUG - Checking if package python-requests is installed.
PyBombs.ReqScanner - DEBUG - Parsing 'requests'
Install tree:
PyBombs.Fetcher.wget - DEBUG - Requirements met.
PyBombs.Fetcher - ERROR - Unexpected error while fetching wget+https://pkgconfig.freedesktop.org/releases/pkg-config-0.28.tar.gz.
PyBombs.Fetcher - ERROR - hostname 'pkgconfig.freedesktop.org' doesn't match either of 'distributions.freedesktop.org', 'farsight.freedesktop.org', 'fontconfig.freedesktop.org', 'fontconfig.org', 'freedesktop.org', 'geoclue.freedesktop.org', 'secure.freedesktop.org', 'www.fontconfig.org', 'www.freedesktop.org'
PyBombs.Packager.source - ERROR - Problem occurred while building package pkg-config:
Unable to fetch recipe pkg-config
PyBombs.install_manager - ERROR - Error installing package pkg-config. Aborting.
```
I'm not even sure this is necessarily a pybombs issue, or in the underlying installers (pip?). I have been unable to find a workaround, so am unable to install uhd or gnuradio using pybombs currently.

drewvian, 2016-05-11:
After much fighting with pybombs I have solved the problem. The problem was with requests (python-requests), not with pybombs itself. I have no clue what caused this to start being a problem now, where it worked fine last week. This link provided the answers I needed.
[https://stackoverflow.com/questions/29099404/ssl-insecureplatform-error-when-using-requests-package/29099439#29099439]

```
sudo apt-get install libffi-dev libssl-dev
sudo -H pip install --upgrade requests
sudo -H pip install --upgrade requests[security]
```
Once these packages were properly handled, I was able to install uhd and gnuradio successfully through pybombs.

I believe maybe some adjustments to the recipes would be good to handle these automatically.

mbr0wn, 2016-05-12:
@drewvian OK, thanks, that was very helpful. However, we also need to fix the recipes such that Ubuntu users don't need to build pkg-config from source.

mbr0wn, 2016-05-13:
I think this particular issue is no longer a problem, but the wget fetcher is definitely not working correctly for some https links. Moving to #331.

closed 2016-05-13 by mbr0wn

marwon34, 2019-01-30:
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyqwt5:
Configuration failed
[ERROR] Error installing package pyqwt5. Aborting.